wpilibsuite/allwpilib #7166: Driver Station loses communication with Romi simulation when integrated with Cheesy Arena FMS

zachb1807, 2024-10-04:
**Description of the Bug**
When using 254’s Cheesy Arena software to run a match with a Romi-based simulation, the Driver Station (DS) loses communication with the Romi simulation as soon as the DS team number is added to the match in Cheesy Arena and the DS registers with the FMS. At this point, no joystick inputs are transferred to the simulation GUI, and the simulation does not receive any updates from the DS or FMS (e.g., alliance changes, match start). Teleop never enables. When the DS is removed from the match, communication is restored.

**Steps to Reproduce**
1. Set up Cheesy Arena FMS on a 10.0.100.x network, with static IP 10.0.100.5.
2. Enable Sim DriverStation in the Romi build file.
3. Use bridge mode to connect the Romi to the network and ensure the Driver Station device is connected as well.
4. Ensure DriverStation connects to the Romi simulation correctly when the FMS is disabled - test that joystick signals from DS are forwarded to the sim GUI.
5. Add the DS’s team number to the match in Cheesy Arena, allowing the DS to register with the FMS.
6. Observe the loss of communication between the DS and Romi simulation.
7. Notice that if the DS team number is removed from the Cheesy Arena match, the DS can communicate with the Romi again and shows teleop disabled.

**Expected Behavior**
- DriverStation should maintain communication with the Romi simulation after the DS registers with the FMS.
- Joystick inputs should be transferred to the simulation, and the simulation should receive updates such as alliance changes and match start.
- Teleop mode should enable when the match begins.

**Screenshots**
[image]
(before the FMS is added: DS works fine with Romi)

[image]
(after DS is connected to the FMS: no robot communication)

[image]
(once the DS is removed from the FMS match: simulation communication restored)

**Desktop:**
 - OS: Windows 10
 - WPILib Information:
Project Version: 2024.1.1
VS Code Version: 1.85.1
WPILib Extension Version: 2024.1.1
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    Romi-Vendordep (1.0.0)
    WPILib-New-Commands (1.0.0)

**Additional context**
- The Romi works fine with Driver Station when the FMS is not involved.
- RoboRio-based robots work fine in the same configuration with Cheesy Arena - the issue only arises with Romi robots.

PeterJohnson, 2024-10-04:
Not connecting to sim when connected to FMS is intentional behavior.  I believe some people have been working on a light FMS specifically for Romi/XRP that connects to WebSockets instead of the traditional control methods.

zachb1807, 2024-10-08:
> Not connecting to sim when connected to FMS is intentional behavior, as we don’t want that to happen accidentally on the field. I believe some people have been working on a light FMS specifically for Romi/XRP that connects to WebSockets instead of the traditional control methods.

Thank you!

closed 2024-10-08 by zachb1807