wpilibsuite/allwpilib #7207: Support for non-critical controllers that don't report unplugged warnings, even when FMS is attached

ArchdukeTim, 2024-10-14:
Especially with the introduction of SysId routines, it's common to have button bindings that are used only on a practice field / workshop. An easy solution for this is to bind to buttons on an unused controller port, and set the test controller to use that port during testing. 

At competition, this means the messages pane is filled with unplugged messages, possible hiding more critical information.

It would be nice if you could mark specific controllers / ports as 'non-critical', and have these warnings hidden if `silenceJoystickWarnings = true`, even when FMS is connected

calcmogul, 2024-10-14:
My concern is teams will use that functionality to disable all joystick warnings for everything. [Alerts](https://github.wpilib.org/allwpilib/docs/development/java/edu/wpi/first/wpilibj/Alert.html) are a better way to declutter the console.

chauser, 2024-10-16:
What we did to avoid this is create a trigger hooked to the presence of the test controller. Only when the controller is plugged in do we bind its buttons, etc. I can share code if you'd like but it's pretty straightforward.

auscompgeek, 2026-10-13:
Joystick connection warnings now use alerts rather than spamming the console.

With opmodes in 2027 I'd recommend using a separate opmode (with scoped triggers in commands v3) if a controller is only considered necessary in certain scenarios. Is there anything else we want to do here?

ThadHouse, 2026-10-13:
Yeah. OpModes are the right solution to this, so I don't think there is anything to do here.

closed 2026-10-13 by ThadHouse