wpilibsuite/allwpilib #7211: [glass] Custom images in Field2d widget do not scale correctly

rmheuer, 2024-10-15:
**Describe the bug**
Custom images in Field2d widget do not scale correctly. The field border and poses appear to be scaled with a fixed padding in pixels from the edge of the window, but the image is scaled with no padding.

**To Reproduce**
Steps to reproduce the behavior:
1. Set the background image of a Field2d widget to a custom image
2. Notice where on the image the yellow border rectangle is
3. Resize the Field2d window
4. The yellow border is over a different part of the image

 - Link to code:
 Doesn't seem that relevant, but here's the code we're using to show the Field2d: https://github.com/SouthwestRoboticsProgramming/2025_Minnetrials/blob/main/Robot/src/main/java/com/swrobotics/robot/logging/FieldView.java

**Expected behavior**
I expected the image to span only the inside of the yellow rectangle or for the yellow rectangle to always be over the same part of the image regardless of size.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint 21.1
 - Project Information: WPILib 2024.3.2, Java OpenJDK 21.0.2

**Additional context**

PeterJohnson, 2024-10-15:
The settings for where the yellow box is placed are settable via the field json file format.  If you just load a jpeg image these are not present and it just draws the box to be 40 pixels from each edge.  Maybe we should remove the yellow box entirely if you just load a jpeg file.

rmheuer, 2024-10-15:
I think the most useful thing would be for the image to be contained in the box, or to have the values from the json file configurable in the GUI for custom images. I like the padding because it lets you still see the whole pose if it's at the origin.

closed 2024-10-16 by PeterJohnson