wpilibsuite/allwpilib #7234: Add a resource ID for Lumyn Labs

nobody5050, 2024-10-20:
Requesting the allocation of a new resource ID for use with upcoming Lumyn Labs products, specifically the ConnectorX breakout board. ConnectorX has two models, each with distinct functions:

1.  **ConnectorX** (General Purpose): A well-rounded board with capabilities for LED control, sensor integration, and power management.
2. **ConnectorX LED**: A model focused primarily on advanced LED control.

The general purpose board is designed to integrate with a variety of daughter boards in different configurations. Because our Product will be reporting the values of its attached sensors, are additional resource IDs required for these sensors or can our Product's ID be the sole value? If accepted, we would appreciate resource types for both the ConnectorX and ConnectorX LED models.

I am also happy to provide technical details or other information necessary to register.

Thanks,
Levi B
Lumyn Labs

rzblue, 2024-11-18:
Could we get more info on what data you're wanting?

To give a bit of context, usage reports are generated from a "resource type" and an "instance number". Instance numbers typically work best as a count of objects of the resource type created, but can alternatively be used to report a secondary type of the resource type for resources that will only be reported once. For example, WPILib tracks creation of PIDController objects using the instance number as a count. However, WPILib reports Odometry usage using the instance number as a subtype (differential, mecanum, swerve), as it's unlikely that a team will use more than one in the same robot.

nobody5050, 2024-12-29:
Hey there, not sure why I didn't get an email from github when you first replied to this, but I'll try to reply now. 

I believe we're looking for the instance number to be used as a signifier for what model of the ConnectorX is being used. While teams may use multiple, it's unlikely and we're looking more to differentiate between which model is used in user code.

sciencewhiz, 2024-12-30:
Unfortunately, it's too late to add usage reporting for 2025

nobody5050, 2024-12-30:
Ah I see. I'll bump it when the time rolls around for 2026

nobody5050, 2025-08-27:
Bumping. I would still be interested in getting a resource type added for the ConnectorX board, and would use the resource number to denote which model.

KangarooKoala, 2025-08-29:
Looks like you're getting a manufacturer ID in #8201, though I'm still not aware of a usage reporting ID being added.
Also, FYI that 2027 will use string-based reporting (see #7763), though of course there's still 2026. (You probably already knew that, but I just wanted to make sure)

nobody5050, 2025-08-29:
I had somehow missed the 2027 changes, thank you for linking to that! I would still like to look into being able to track usage in the 2026 season though