wpilibsuite/allwpilib #7239: BatterySim::Calculate can return negative battery voltage

mcm001, 2024-10-21:
**Describe the bug**

BatterySim::Calculate can return negative battery voltage if sufficient current draw is requested. 

The [C++ impl](https://github.com/wpilibsuite/allwpilib/blob/5ac132f6a250db8539a572192dba530d61a4d1e6/wpilibc/src/main/native/include/frc/simulation/BatterySim.h#L33) just takes 12_v - (i * R) unbounded

The [Java impl](https://github.com/wpilibsuite/allwpilib/blob/5ac132f6a250db8539a572192dba530d61a4d1e6/wpilibj/src/main/java/edu/wpi/first/wpilibj/simulation/BatterySim.java#L28) does the same.

mcm001, 2024-10-21:
Something more physically accurate, such as https://www.mathworks.com/help/simscape/ug/lead-acid-battery.html , might be considered.

calcmogul, 2024-10-21:
The easiest fix for now is clamping the return value to >= 0.

calcmogul, 2024-10-22:
That assumes our battery sim is completely accurate, which it's not at super high current loads. We also don't model current limits.

PR #7325 (2024-11-03): [simulation] Clamp battery voltage to 0
Resolves #7239
closed 2024-11-03 by PeterJohnson