wpilibsuite/allwpilib #7241: imgui libraries are not published

Gold856, 2024-10-21:
We only publish headers and source. I'm not actually sure if we need to publish those libraries (are there any downstream consumers?)

virtuald, 2024-10-21:
I would like to wrap the imgui/wpigui combination in python at some point, so I'll probably complain about that when I get to it.

There are various imgui python wrappers, but many are unmaintained or are out of date or only support some platforms.

PeterJohnson, 2024-10-21:
I thought we statically linked the imgui libraries into wpigui--is that not the case?

Gold856, 2024-10-21:
I checked the wpigui libraries, and we do statically link imgui into wpigui. It's just that the headers and source for imgui are located elsewhere `edu/wpi/first/thirdparty/imguiSuite/imguiSuite-cpp`. It's a bit confusing though to have the headers split across two different places.

PR #7575 (2024-12-24): [build] Fix imgui libraries not being published
Fixes mismatched library name between imgui_suite's build.gradle and publish.gradle by renaming the library to imguiSuite.
Fixes #7241.
closed 2024-12-24 by PeterJohnson