wpilibsuite/allwpilib #7275: Make NumericalJacobian support dynamic matrix sizes

calcmogul, 2024-10-23:


LucienMorey, 2025-05-04:
Will the Jacobian being computed always be square here? The template args would imply that there was supposed to be intent to do things with a rectangular Jacobian somehow, but the function arg for `x` is a column vector with `Cols` number of rows, implying the output type will be square, right?

calcmogul, 2025-05-04:
Per the docs, `Rows` is number of rows in the Jacobian matrix and `Cols` is the number of columns in the Jacobian matrix. The confusion is due to the Jacobian being transposed.

LucienMorey, 2025-05-04:
I'm happy to take you at your word on this one. I will have one more try at explaining my confusion before just trying to do the bits. I have a little experience with nonlinear control, so I know you can do the things if you handle it properly. My question was around this implementation. I would expect some erroneous behaviour, likely a C++ exception rather than mathematical craziness if the function ever had to handle a non-square Jacobian.

If `x` is a column vector of `Cols` rows and the result matrix has `Rows` rows, the column-by-column assignment of the computed partial derivative will fail if those template args aren't equal.  I think the function works as you suggest if the supplied argument `x` has `Rows` rows rather than `Cols` rows. The iteration can then be done for the number of columns like it is now without a mismatch.

KangarooKoala, 2025-05-05:
If I'm understanding things right, `f` is a function from `Vectord<Cols>` to `Vectord<Rows>`, so `x`, `dX_plus`, and `dX_minus` are have `Cols` rows since they're inputs but the resultant matrix has `Rows` rows since that's how many rows the output vectors have.

LucienMorey, 2025-05-05:
OK. That's the bit that I wasn't fully thinking through. This holds up if the dynamics function can output a different-sized vector to the supplied input.

Now to work on the actual issue! I was hoping to assume something square to size the resultant matrix from the input vector based on the `.rows()` function. I think this means I have to peer into the magic of `decltype` to try and peer into the return of `f`.

LucienMorey, 2025-05-05:
Lol no I didnt. Pr submitted!

PR #7959 (2025-05-06): Add dynamic size support for numerical jacobian computation
Closes #7275
closed 2025-05-06 by PeterJohnson