wpilibsuite/allwpilib #7357: Make C++ RomiGyro and XRPGyro return SI unit types

calcmogul, 2024-11-06:
The angle getters should return radians and the angular rate getters should return radians per second.

Advay17, 2024-11-07:
I'll get this one done in a few hours.

Advay17, 2024-11-07:
Should the old methods that return degrees be deprecated? If so, should they be renamed and how should they be renamed?

Advay17, 2024-11-07:
In addition, should another issue be opened to update Java to use radians instead of degrees?

Advay17, 2024-11-07:
On digging into the code, I realized that the servo library also needs to be changed to use SI unit types. However, it has two sets of methods: one to get/set the angle in degrees(getAngle and setAngle), and the other to get/set the angle in radians(getPosition and setPosition). Since conversion would be easier using the units library, should the former set of methods be deprecated and the latter changed to return units::radians_t?

calcmogul, 2024-11-07:
> However, it has two sets of methods: one to get/set the angle in degrees(getAngle and setAngle), and the other to get/set the angle in radians(getPosition and setPosition).

`setPosition(double)` takes half-rotations, not radians. The angle in degrees is divided by 180 to produce a value in the range [0, 1].

> should the former set of methods be deprecated and the latter changed to return units::radians_t?

We should just have a SetAngle() function that takes radians, with everything else deprecated.

PR #7366 (2024-11-09): Update C++ XRP Code to use SI Units
Closes #7357
closed 2024-11-09 by PeterJohnson