wpilibsuite/allwpilib #7382: [ntcore] add `UnitTopic`

spacey-sooty, 2024-11-12:
This would be a topic that stores a double with unit information represented in NT metadata

spacey-sooty, 2024-11-15:
Maybe something that could be included in wpilibc/j so it's still easily accessible to teams?

Starlight220, 2024-11-20:
The idea would be a library that takes a `Supplier<Measure<?>>` etc, rather than users delving into unit-typed Topic objects etc

spacey-sooty, 2024-11-21:
> The idea would be a library that takes a Supplier<Measure<?>> etc, rather than users delving into unit-typed Topic objects etc

I actually disagree with this. A unit typed topic which behaves the same as other topics makes sense as an abstraction here IMO.

> Wouldn't that still require modifying NetworkTableInstance to add getUnitTopic(), making it still depend on the units library? Unless we want teams to instead use getDoubleTopic() for UnitTopic, which would be inconsistent

You should be able to create this topic by injecting the instance or parent table instead of using this which would avoid modifying ntcore