wpilibsuite/allwpilib #7393: Overload  PenumaticsBase.getForType

Isaac-dot-py, 2024-11-15:
I'm always frustrated when I find spaghetti code in a library. In this specific case than Compressor has to call PenumaticsBase.getDefaultForType just to use the right constructor for PenumaticsBase.

PenumaticsBase.getForType should be overloaded so that you Compressor can call PenumaticsBase.getForType(PneumaticsModuleType type), which automatically calls PenumaticsBase.getDefaultForType.

ThadHouse, 2024-11-15:
Those functions are static. You can't overload statics.

ThadHouse, 2024-11-15:
I'm not fully sure how that would clean anything up. getForType and getDefaultForType return very different things. What would you overload?

The only thing I could envision would be adding 3rd constructor to Compressor that takes in a already constructed PneumaticsBase. But adding overloaded constructors is definitely more spaghetti than this is.

closed 2024-11-15 by Isaac-dot-py