wpilibsuite/allwpilib #7399: [ntcore] setRetained doesn't work

spacey-sooty, 2024-11-17:
**Describe the bug**
When I use the function setRetained on a topic in Java it doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/spacey-sooty/nt-set-retained-repro
2. Run in simulation
3. Check glass to see topic Foo doesn't exist under retained[image]. 
I don't think this is a glass issue because when I disconnect and reconnect in AdvantageScope the values of any subtopics go away.

**Expected behavior**
The Topic should have the retained property. This would cause glass to display it under the retained section and AdvantageScope to keep it between disconnects.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04
 - WPILib Information:
Project Version: 2025.1.1-beta-1
VS Code Version: 1.94.2
WPILib Extension Version: 2025.1.1-beta-1
C++ Extension Version: 1.22.9
Java Extension Version: 1.36.2024092708
Java Debug Extension Version: 0.58.2024090204
Java Dependencies Extension Version 0.24.0
Java Version: 17
Java Location: /home/jadey/wpilib/2025/jdk
Vendor Libraries:
    WPILib-New-Commands (1.0.0)

PeterJohnson, 2024-11-17:
The reason for this is the publish() overwrites the properties since the topic had not yet been published.  If you move the setretained to after the publish or pass the retained as part of the publish properties parameter, this will work.  I agree this is confusing behavior that should be changed (or at least documented).  On the server side, property sets to non-existent topics are ignored, but agreed this is confusing in an API sense.

If we are changing this behavior, we need to redefine what the properties parameter to publish() means.  Currently what it is defined to be is the “initial” properties for a topic—the properties that are set if the topic does not already exist.  We could change it to a “merge” behavior with previously locally set properties only *if* the topic didn’t already exist? Note this can still react in an unpredictable way if the topic gets published by someone else in between the setRetained() and publish() calls, so the “right” way will still be to set initial properties through the publish() call directly, but it will work in the likely common case.  Not sure if that’s better or worse though.

spacey-sooty, 2024-11-18:
> so the “right” way will still be to set initial properties through the publish() call directly
I'm probably missing something but when I look at PubSubOption I don't see a way to set retained?

spacey-sooty, 2024-11-18:
Ah that explains why I didn't see it... yeah we should add a better way to that I'll make an issue.

spacey-sooty, 2024-11-18:
Hm interestingly even when I set that property my struct publisher doesn't persist across disconnects in AdvantageScope but I can only reproduce it with long arrays...