wpilibsuite/allwpilib #7412: Foonathan memory library is missing generated header

Gold856, 2024-11-19:
If you open up `container_node_sizes.hpp`, you'll notice that it includes `container_node_sizes_impl.hpp`. Looking at the foonathan memory repo, it appears this header is generated and might be machine specific. This header is needed to use any of the custom containers, and our build currently does not generate it. We should figure out how to get our Gradle build to generate this header.

Advay17, 2024-11-20:
It appears that it is generated using a cmake step:

```
# generate container_node_sizes.hpp if necessary
if(NOT EXISTS ${CMAKE_CURRENT_BINARY_DIR}/container_node_sizes_impl.hpp)
    get_container_node_sizes(${CMAKE_CURRENT_BINARY_DIR}/container_node_sizes_impl.hpp)
endif()`

From:[here](https://github.com/foonathan/memory/blob/016c9fbd61b57ed2058551dcf225c15a0e716cce/src/CMakeLists.txt#L80)

Advay17, 2024-11-20:
It appears that it is generated using a cmake step:

`if(NOT EXISTS ${CMAKE_CURRENT_BINARY_DIR}/container_node_sizes_impl.hpp)`
    `get_container_node_sizes(${CMAKE_CURRENT_BINARY_DIR}/container_node_sizes_impl.hpp)`
`endif()`

From:[here](https://github.com/foonathan/memory/blob/016c9fbd61b57ed2058551dcf225c15a0e716cce/src/CMakeLists.txt#L80)

PeterJohnson, 2024-11-20:
That works for cmake, but we'll need to replicate the logic (or work around it, e.g. with ifdefs for our specific supported platforms) for other build systems (bazel, Gradle).

Advay17, 2024-11-21:
I believe I have the cmake portion completed, in what folder should the generated file go?

sciencewhiz, 2025-07-12:
Foonathan memory was removed in #8035

closed 2025-07-12 by sciencewhiz