wpilibsuite/allwpilib #7413: Make Command-Based Trigger.whileTrue, onTrue, whileFalse, onFalse true to their names

tom131313, 2024-11-20:
Trigger the command on condition of changing state (the current action) AND on the initial state (the new, improved action).

For example, `whileTrue` would trigger when the condition changes from false to true. In addition, the proposed new trigger activates if the initial state of the condition is true. These actions would happen if the command can be scheduled - that is the mode is Enabled or the command can run in Disabled.

This has the potential of being a breaking change for some users. I suggest methods with both behaviors could be supported.

The current motivation for this change is a bumper limit switch on a robot should trigger a command that prevents certain movement in that state. When that command is used to move away from an obstacle and not toward the obstacle the limited command ends and the normal default command resumes. Because the current trigger behavior doesn't honor the initial state, the work-around is tedious to use triggers to change commands for the change of state.
```java
  /**
   * Starts the given command when the condition changes to `true` or is initially `true`
   * and cancels it when the condition changes to `false`.
   *
   * <p>
   * Doesn't re-start the command if it ends while the condition is still `true`.
   * If the command should restart, see {@link edu.wpi.first.wpilibj2.command.RepeatCommand}.
   *
   * @param command the command to start
   * @return this trigger, so calls can be chained
   */
  public Trigger whileTrue(Command command) {
    requireNonNullParam(command, "command", "whileTrue");
    m_loop.bind(
	
        new Runnable() {
          private boolean m_pressedLast; // initial value ignored - protected by firstTime
          private boolean firstTime = true; // indicates the command has never been scheduled
		  
          @Override
          public void run() {
            boolean pressed = m_condition.getAsBoolean();

            if ((command.runsWhenDisabled() || DriverStation.isEnabled()) // command can be scheduled
                && pressed  // desired condition
                && ((!firstTime && !m_pressedLast) // state has changed
                    || firstTime)) { // initially suppress comparison to previous state
                  firstTime = false;
                  command.schedule();
              }

            else if (!firstTime && m_pressedLast && !pressed) { // has state reverted
                command.cancel();
            }

            m_pressedLast = pressed;
          }
        });
		
    return this;
  }
```

rzblue, 2024-11-20:
Duplicate of #5608

tom131313, 2024-11-20:
#5608 is a special case of this more general description and solution. I searched the issues but missed that one. I'm sorry I didn't see it. Thanks for your better memory to find the very similar issue.