wpilibsuite/allwpilib #7448: DCMotorSim acceleration (a bug of my own making?)

narmstro2020, 2024-11-29:
In the the getAngularAccelerationRadPerSecSq() function a matrix named acceleration is calculated from Ax + Bu.  This would result in xdot.  
The returned result is the first row of the xdot matrix, which would be velocity.  Shouldn't it be  the second row at index 1 for acceleration.  

I know I wrote code to this effect in a PR.  

Also is there a way to use the discretized versions of A and B instead of the continuous form just out of curiosity.

calcmogul, 2024-11-29:
Yep, it should be index 1.

WPILib uses continuous models everywhere because they're easier to formulate. We have functions for discretizing them internally.

narmstro2020, 2024-11-29:
Excellent.  I'll make a PR fixing this.  
I've been tooling around with my own versions of the sims using the beta code.  Mostly just modifications to the plumbing and further usages of mutable unit class elements when I saw this one.  

You probably know that I've got a torque/current and voltage sim PR in the works.  

Some of that plumbing exists in my little experiments on the side.  

I haven't done any updates on that PR in a while due to lack of feedback and my own scheduling/timing issues.  

I made another one recently that just does a reorg on DCMotor and introduces a Gearbox class.  These components are in the other PR as well.   

I'm sure the lack of feedback involves the sim being a lower priority as we approach the 2025 season and not a ("Ooh that's bad and I don't know how to put it nicely, haha)

PR #7449 (2024-11-30): [wpimath] fix to acceleration getter for DCMotorSim
The matrix variable was extracting the wrong row. It should be row index 1 and was row index 0.

Fixes #7448.
closed 2024-11-30 by PeterJohnson