wpilibsuite/allwpilib #7477: ChassisSpeed.fromFieldRelavtive deprecated, suggested replacement is a void function.

JR-Wiederkehr, 2024-12-02:
The fromRelativeSpeed function that returns a ChassisSpeed object is now deprecated. The suggested replacement, toRobotRelative, is a void function that returns nothing. Whats more confusing is that it pulls the x and y speeds out of thin air, as there is no included way to pass these speeds into the function.

How are we to use this void function to replace the functionality of fromRelativeSpeed?

PeterJohnson, 2024-12-02:
The toRobotRelativeSpeeds function mutates the ChassisSpeeds object in place.  The speeds come from the ChassisSpeeds object itself and the object is updated with the new speeds.

JR-Wiederkehr, 2024-12-02:
Got some help from the discord and figured it out. Created a pull Request on Rev Robotics MAX_Swerve template to update to handle the deprecation. 

Thank you

closed 2024-12-02 by JR-Wiederkehr