wpilibsuite/allwpilib #7483: SimpleMotorFeedforward : Another bug of my own making?

narmstro2020, 2024-12-04:
In the calculate overload 

`  public <U extends Unit> Voltage calculate(
      Measure<? extends PerUnit<U, TimeUnit>> currentVelocity,
      Measure<? extends PerUnit<U, TimeUnit>> nextVelocity) {
    // See wpimath/algorithms.md#Simple_motor_feedforward for derivation
    if (ka == 0.0) {
      return Volts.of(ks * Math.signum(nextVelocity.magnitude()) + kv * nextVelocity.magnitude());
    } else {
      double A = -kv / ka;
      double B = 1.0 / ka;
      double A_d = Math.exp(A * m_dt);
      double B_d = 1.0 / A * (A_d - 1.0) * B;
      return Volts.of(
          ks * Math.signum(currentVelocity.magnitude())
              + 1.0 / B_d * (nextVelocity.magnitude() - A_d * currentVelocity.magnitude()));
    }
  }`

Shouldn't the Math.signum in the if (ka == 0.0) block be currentVelocity not nextVelocity?

calcmogul, 2024-12-04:
No, as per the derivation in [wpimath/algorithms.md](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/algorithms.md).
```
uₖ = kₛ sgn(x) + kᵥxₖ₊₁
```

narmstro2020, 2024-12-04:
Just to clarify x not x_k because of the line where we keep sgn(x) constants for the duration of the step?

calcmogul, 2024-12-04:
If the sign stays the same, it doesn't matter whether xₖ or xₖ₊₁ is used. If the sign changes during a timestep, one could argue for using either.