wpilibsuite/allwpilib #7493: [example] Mecanum controller command example has incorrect wheel order

KangarooKoala, 2024-12-06:
**Describe the bug**
In the drive subsystem of the mecanum controller command example, the order that the wheels are passed to the MecanumDriveWheelSpeeds and MecanumDriveWheelPositions constructors is wrong. It uses frontLeft, rearLeft, frontRight, rearRight, when the actual constructors use frontLeft, frontRight, rearLeft, rearRight.

**Links to code**
 - Java: https://github.com/wpilibsuite/allwpilib/blob/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/mecanumcontrollercommand/subsystems/DriveSubsystem.java
 - C++: https://github.com/wpilibsuite/allwpilib/blob/main/wpilibcExamples/src/main/cpp/examples/MecanumControllerCommand/cpp/subsystems/DriveSubsystem.cpp

**Expected behavior**
The order in the example should be correct. (That is, the example should use frontLeft, frontRight, rearLeft, and rearRight)

**Additional context**
This would be a relatively low effort issue- Just swapping the order of two arguments in 4 places and double checking the other examples for any other bugs.

Advay17, 2025-04-21:
Are you sure that the order is incorrect? Is the constructor not [`public MecanumDrive(
      MotorController frontLeftMotor,
      MotorController rearLeftMotor,
      MotorController frontRightMotor,
      MotorController rearRightMotor)`](https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java#L119)?

KangarooKoala, 2025-04-22:
The order correctly matches the `MecanumDrive` constructor, but it doesn't not match the `MecanumDriveWheelSpeeds` and `MecanumDriveWheelPositions` constructors, which are the ones described in the issue.

Conflicting orders:
* [`MecanumDrive(frontLeft, rearLeft, frontRight, rearRight)`](https://github.com/wpilibsuite/allwpilib/blob/26e299115fab948178d0e884af6a2c4d517ae4a3/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java#L146)
* [`MecanumDrive.State(frontLeft, frontRight, rearLeft, rearRight)`](https://github.com/wpilibsuite/allwpilib/blob/26e299115fab948178d0e884af6a2c4d517ae4a3/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java#L100)
* [`MecanumDriveWheelSpeeds(frontLeft, frontRight, rearLeft, rearRight)`](https://github.com/wpilibsuite/allwpilib/blob/26e299115fab948178d0e884af6a2c4d517ae4a3/wpimath/src/main/java/edu/wpi/first/math/kinematics/MecanumDriveWheelSpeeds.java#L46)
* [`MecanumDriveWheelPositions(frontLeft, frontRight, rearLeft, rearRight)`](https://github.com/wpilibsuite/allwpilib/blob/26e299115fab948178d0e884af6a2c4d517ae4a3/wpimath/src/main/java/edu/wpi/first/math/kinematics/MecanumDriveWheelPositions.java#L53)

The example consistently uses frontLeft, rearLeft, frontRight, rearRight, which is only correct for `MecanumDrive`.

Advay17, 2025-04-22:
Ah, I see the issue. I feel like this discrepancy would be the cause of a lot of similar issues for teams, where they potentially write the wrong order of the wheels. Is there a reason for this difference, and if not, would it be more beneficial to modify the constructor itself as well to prevent future similar minor mistakes. Or perhaps something can done with a functional interface MecanumWheel with a generic that takes in an enum.