wpilibsuite/allwpilib #7494: Installer crashes on Linux - cannot launch

heyallnorahere, 2024-12-06:
The WPILib installer crashes on Linux due to an issue in Skia via Avalonia

**To Reproduce**
1. Download the `2025.1.1-beta-2` installer version for x64 Linux
2. Extract the archive
3. Run the installer in the extracted directory

**Log**
```
$ ./WPILibInstaller 
Unhandled exception. System.InvalidOperationException: Default font family name can't be null or empty.
   at Avalonia.Media.FontManager..ctor(IFontManagerImpl platformImpl) in /_/src/Avalonia.Visuals/Media/FontManager.cs:line 28
   at Avalonia.Media.FontManager.get_Current() in /_/src/Avalonia.Visuals/Media/FontManager.cs:line 48
   at Avalonia.Media.Typeface.get_GlyphTypeface() in /_/src/Avalonia.Visuals/Media/Typeface.cs:line 69
   at Avalonia.Media.TextFormatting.TextCharacters.TryGetRunProperties(ReadOnlySlice`1 text, Typeface typeface, Typeface defaultTypeface, Int32& count) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextCharacters.cs:line 132
   at Avalonia.Media.TextFormatting.TextCharacters.CreateShapeableRun(ReadOnlySlice`1 text, TextRunProperties defaultProperties) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextCharacters.cs:line 62
   at Avalonia.Media.TextFormatting.TextCharacters.GetShapeableCharacters() in /_/src/Avalonia.Visuals/Media/TextFormatting/TextCharacters.cs:line 40
   at Avalonia.Media.TextFormatting.TextFormatterImpl.FetchTextRuns(ITextSource textSource, Int32 firstTextSourceIndex, TextLineBreak previousLineBreak, TextLineBreak& nextLineBreak) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextFormatterImpl.cs:line 286
   at Avalonia.Media.TextFormatting.TextFormatterImpl.FormatLine(ITextSource textSource, Int32 firstTextSourceIndex, Double paragraphWidth, TextParagraphProperties paragraphProperties, TextLineBreak previousLineBreak) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextFormatterImpl.cs:line 15
   at Avalonia.Media.TextFormatting.TextLayout.UpdateLayout() in /_/src/Avalonia.Visuals/Media/TextFormatting/TextLayout.cs:line 225
   at Avalonia.Media.TextFormatting.TextLayout..ctor(String text, Typeface typeface, Double fontSize, IBrush foreground, TextAlignment textAlignment, TextWrapping textWrapping, TextTrimming textTrimming, TextDecorationCollection textDecorations, Double maxWidth, Double maxHeight, Double lineHeight, Int32 maxLines, IReadOnlyList`1 textStyleOverrides) in /_/src/Avalonia.Visuals/Media/TextFormatting/TextLayout.cs:line 71
   at Avalonia.Controls.TextBlock.CreateTextLayout(Size constraint, String text) in /_/src/Avalonia.Controls/TextBlock.cs:line 475
   at Avalonia.Controls.Primitives.AccessText.CreateTextLayout(Size constraint, String text) in /_/src/Avalonia.Controls/Primitives/AccessText.cs:line 156
   at Avalonia.Controls.TextBlock.get_TextLayout() in /_/src/Avalonia.Controls/TextBlock.cs:line 166
   at Avalonia.Controls.TextBlock.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/TextBlock.cs:line 521
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding, Thickness borderThickness) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 39
   at Avalonia.Controls.Presenters.ContentPresenter.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Presenters/ContentPresenter.cs:line 366
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.Layoutable.MeasureOverride(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 625
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Controls.Grid.MeasureCell(Int32 cell, Boolean forceInfinityV) in /_/src/Avalonia.Controls/Grid.cs:line 1150
   at Avalonia.Controls.Grid.MeasureCellsGroup(Int32 cellsHead, Size referenceSize, Boolean ignoreDesiredSizeU, Boolean forceInfinityV, Boolean& hasDesiredSizeUChanged) in /_/src/Avalonia.Controls/Grid.cs:line 1005
   at Avalonia.Controls.Grid.MeasureCellsGroup(Int32 cellsHead, Size referenceSize, Boolean ignoreDesiredSizeU, Boolean forceInfinityV) in /_/src/Avalonia.Controls/Grid.cs:line 968
   at Avalonia.Controls.Grid.MeasureOverride(Size constraint) in /_/src/Avalonia.Controls/Grid.cs:line 448
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding, Thickness borderThickness) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 39
   at Avalonia.Controls.Presenters.ContentPresenter.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Presenters/ContentPresenter.cs:line 366
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutHelper.MeasureChild(ILayoutable control, Size availableSize, Thickness padding) in /_/src/Avalonia.Layout/LayoutHelper.cs:line 46
   at Avalonia.Controls.Decorator.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Decorator.cs:line 54
   at Avalonia.Controls.Primitives.VisualLayerManager.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Primitives/VisualLayerManager.cs:line 133
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.Layoutable.MeasureOverride(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 625
   at Avalonia.Layout.Layoutable.MeasureCore(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 559
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.Layoutable.MeasureOverride(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 625
   at Avalonia.Controls.Window.MeasureOverride(Size availableSize) in /_/src/Avalonia.Controls/Window.cs:line 888
   at Avalonia.Controls.WindowBase.MeasureCore(Size availableSize) in /_/src/Avalonia.Controls/WindowBase.cs:line 237
   at Avalonia.Layout.Layoutable.Measure(Size availableSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 364
   at Avalonia.Layout.LayoutManager.Measure(ILayoutable control) in /_/src/Avalonia.Layout/LayoutManager.cs:line 287
   at Avalonia.Layout.LayoutManager.ExecuteInitialLayoutPass() in /_/src/Avalonia.Layout/LayoutManager.cs:line 174
   at Avalonia.Controls.Window.ShowCore(Window parent) in /_/src/Avalonia.Controls/Window.cs:line 688
   at Avalonia.Controls.Window.Show() in /_/src/Avalonia.Controls/Window.cs:line 626
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 125
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 175
   at WPILibInstaller.Program.Main(String[] args) in /home/runner/work/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/Program.cs:line 11
Aborted (core dumped)
```

**Expected behavior**
To install WPILib 2025.1.1 Beta 2.

**Desktop**
 - OS: Ubuntu 24.10
 - Project Information: N/A

**Additional context**
On Linux, Avalonia [calls to Skia](https://github.com/AvaloniaUI/Avalonia/blob/master/src/Skia/Avalonia.Skia/FontManagerImpl.cs#L17) to retrieve the default font family for the system. In an older version of Skia (and I suspect the version this installer uses), the Linux implementation of this call returns a nullptr string. This issue seems to have been fixed in [this commit](https://github.com/google/skia/commit/25272300c308335f2d87c2a70e249965b268bcb0#diff-b017452ef0415fa062aeefa11bad666b28b442c8ac1b51b71b54dd841a0cb27bR901).

heyallnorahere, 2024-12-06:
Additional context:
Completely configured locale. I do not believe it is an issue with locale settings.
```
$ locale
LANG=en_US.UTF-8
LANGUAGE=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```

PeterJohnson, 2024-12-06:
See https://github.com/wpilibsuite/WPILibInstaller-Avalonia/issues/153

heyallnorahere, 2024-12-06:
My bad, didn't see that.

closed 2024-12-06 by heyallnorahere

sciencewhiz, 2025-03-28:
Can you try this installer and see if it works without using the workaround? It updates Avalonia and Skia. https://github.com/wpilibsuite/WPILibInstaller-Avalonia/actions/runs/14051034586/artifacts/2812852446