wpilibsuite/allwpilib #7559: Accept custom gravity strength in elevator and arm simulators

calcmogul, 2024-12-17:
This allows simulating tilted elevators and arms, including ones that translate/rotate in the x-y plane.

narmstro2020, 2024-12-18:
Wouldn't hurt to allow for arm MOI around different pivot locations.

calcmogul, 2024-12-18:
The arm pivot doesn't move though.

narmstro2020, 2024-12-18:
Oh I’m sorry I meant on system construction.  Right now it assume the pivot is at the exact opposite end

Advay17, 2025-04-21:
Should this be a value set in the constructor? If so, should it replace the `simulateGravity` boolean already present(given that if we want a different value for gravity, we would probably want to simulate it)?

calcmogul, 2025-04-21:
Yea, we can just replace the bool with a double for gravitational acceleration, with an API doc note about how users should use it for tilted mechanisms (e.g., g cosθ).

narmstro2020, 2025-04-22:
I've noticed that kg / ka from sysId is significantly lower than 9.8 when done in the right units.  Is this because the model is only assuming mr^2 for the moment of inertia and not the inertia of the elevator's gearbox itself?  I originally thought frictional loses, but that should be acounted for by ks?