wpilibsuite/allwpilib #7589: I am unanble to build robot code after I updated to Beta-3

kglasnapp, 2024-12-25:
The error I get is:

* What went wrong:
An exception occurred applying plugin request [id: 'edu.wpi.first.GradleRIO', version: '2025.1.1-beta-3']
> Failed to apply plugin class 'edu.wpi.first.gradlerio.wpi.WPIPlugin'.
   > Could not create an instance of type edu.wpi.first.gradlerio.wpi.WPIExtension.
      > Conflicting Vendor Dependency 60b2694b-9e6e-4026-81ee-6f167946f4b0 for uuid 1ccce5a4-acd2-4d18-bca3-4b8047188400. Reason: ThriftyLib is required!

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 3s

The first few lines in build.gradle are:

plugins {
    id "java"
    id("edu.wpi.first.GradleRIO") version "2025.1.1-beta-3"
}

java {
    sourceCompatibility = JavaVersion.VERSION_17
    targetCompatibility = JavaVersion.VERSION_17
}

def ROBOT_MAIN_CLASS = "frc.robot.Main"

Help

PeterJohnson, 2024-12-25:
What other vendor libraries are you using?  Most likely there's a vendor library (YAGSL?) that requires the ThriftyLib vendor dependency be installed, and it isn't, thus the error.

rzblue, 2024-12-25:
This doesn't appear to be related to updating to 2025.1.1-beta-3. Based on the error message, you're using yagsl and do not have ThriftyLib installed. Installing ThriftyLib should resolve this.

kglasnapp, 2024-12-25:
Thanks so very much -- I removed yagsl and I am able to build code.  I also put in https://docs.home.thethriftybot.com/ThriftyLib.json and I can compile with yagsl.

Keith

closed 2024-12-25 by kglasnapp