wpilibsuite/allwpilib #7597: Wpilib installer showing transparent screen and not responding

MiracleAig, 2024-12-27:
**Describe the bug**
The wpilib installer window shows up transparent and unresponsive 

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest wpilib version and mount the iso
2. Open the installer

**Expected behavior**
The installer window opens but is see through and does not respond, I have to manually end the task for the window to close 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 home 24H2, OS build 26100.2605
 - Project Information: Wpilib version 2024.3.2

Gold856, 2024-12-27:
Can't reproduce. Can you try the latest 2025 beta? https://github.com/wpilibsuite/allwpilib/releases/tag/v2025.1.1-beta-3

MiracleAig, 2024-12-27:
I did I experienced the same issue

Gold856, 2024-12-27:
Try disabling any antivirus software you might have. Also, are you running normal Windows 11 Home and not an N/KN edition?