wpilibsuite/allwpilib #7598: Simulator closed instantly after appeard without bug report

RockyXRQ, 2024-12-27:
**Describe the bug**
Same project can be simulated normally in intellij Idea, but in VSCode the window of simulator closed instantly after appear without any bug report

**To Reproduce**
Steps to reproduce the behavior:
1. run normally in Idea with: (auto generated by FRC plugin of intellij)
```xml
<component name="ProjectRunConfigurationManager">
  <configuration default="false" name="Build &amp; Run Simulate Java" type="GradleRunConfiguration" factoryName="Gradle">
    <ExternalSystemSettings>
      <option name="executionName" />
      <option name="externalProjectPath" value="$PROJECT_DIR$" />
      <option name="externalSystemIdString" value="GRADLE" />
      <option name="scriptParameters" />
      <option name="taskDescriptions">
        <list />
      </option>
      <option name="taskNames">
        <list>
          <option value="simulateJava" />
        </list>
      </option>
      <option name="vmOptions" />
    </ExternalSystemSettings>
    <ExternalSystemDebugServerProcess>true</ExternalSystemDebugServerProcess>
    <ExternalSystemReattachDebugProcess>true</ExternalSystemReattachDebugProcess>
    <DebugAllEnabled>false</DebugAllEnabled>
    <RunAsTest>false</RunAsTest>
    <method v="2" />
  </configuration>
  <configuration default="false" name="Build &amp; Run Simulate Java" type="GradleRunConfiguration" factoryName="Gradle">
    <ExternalSystemSettings>
      <option name="executionName" />
      <option name="externalProjectPath" value="$PROJECT_DIR$" />
      <option name="externalSystemIdString" value="GRADLE" />
      <option name="scriptParameters" />
      <option name="taskDescriptions">
        <list />
      </option>
      <option name="taskNames">
        <list>
          <option value="simulateJava" />
        </list>
      </option>
      <option name="vmOptions" />
    </ExternalSystemSettings>
    <ExternalSystemDebugServerProcess>true</ExternalSystemDebugServerProcess>
    <ExternalSystemReattachDebugProcess>true</ExternalSystemReattachDebugProcess>
    <DebugAllEnabled>false</DebugAllEnabled>
    <RunAsTest>false</RunAsTest>
    <method v="2" />
  </configuration>
</component>
```

2. closed instantly after simuator window appeard without any bug report
[image]

**Expected behavior**
simulator keep running until user to shutdown it or closed with bug report

**Desktop:**
 - OS: Windows 10 x64
 - Project Information: 
WPILib Information:
Project Version: 2024.3.2
VS Code Version: 1.93.1
WPILib Extension Version: 2025.1.1-beta-3
C++ Extension Version: Not Installed
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    AdvantageKit (3.2.1)
    CTRE-Phoenix (v5) (5.33.1)
    CTRE-Phoenix (v6) (24.3.0)
    photonlib (v2024.3.1)
    WPILib-New-Commands (1.0.0)

**Additional context**
I tried several new created project with wpilib-2025-beta-3, but all failed with simulation. I can run them with beta2, but none of then can run with beta3

ThadHouse, 2024-12-27:
Something is weird with that information window. It’s showing the project version is 2024, but using the 2025 wpilib install, along with the c++ extension missing. Are you sure you ran the 2025 beta installer, and created the project using 2025 vs code?

RockyXRQ, 2024-12-27:
> Something is weird with that information window. It’s showing the project version is 2024, but using the 2025 wpilib install, along with the c++ extension missing. Are you sure you ran the 2025 beta installer, and created the project using 2025 vs code?

Sorry Thad I updated the wrong project info! now it should be correct!

RockyXRQ, 2024-12-27:
> Something is weird with that information window. It’s showing the project version is 2024, but using the 2025 wpilib install, along with the c++ extension missing. Are you sure you ran the 2025 beta installer, and created the project using 2025 vs code?

I use the 2025 beta installer, but this is a old project generated with Intellij Idea, I tried to run simulator with 2025 beta installer and also Cursor with a custom installed wpilib plugin through vsix, and they all have the same result

rzblue, 2024-12-27:
Have you tried a project created with the 2025.1.1-beta-3 vscode plugin?

RockyXRQ, 2024-12-28:
> Have you tried a project created with the 2025.1.1-beta-3 vscode plugin?

seems a new created project doesn't have this issue. I will reopen it if there's more problem. Thx for the helping!!!

closed 2024-12-28 by RockyXRQ