wpilibsuite/allwpilib #7603: [upstream_utils] Can't run on Windows

WispySparks, 2024-12-28:
Upstream utils scripts can't be run on windows due to many paths having hardcoded `/`. Could be fixed by replacing these with os.path.join or os.path.sep.

Gold856, 2024-12-29:
This is a problem for only a select few scripts that have direct comparisons between raw paths specified by hand, like `./imgui.h` and paths created with `os.path.join`. The two paths will have inconsistencies in terms of slashes, and the comparisons will therefore fail (because they don't match). One solution is to use `os.path.join` as a kind of "input sanitizer", and preprocess all manually specified paths with `os.path.join` before passing them to functions.

auscompgeek, 2024-12-29:
It'd be worth fixing these scripts to use pathlib instead.

calcmogul, 2024-12-29:
Although pathlib doesn't add `pathlib.Path.walk()` (an equivalent of os.walk()) until Python 3.12, and we support down to Python 3.9 at the moment.

TheTripleV, 2024-12-29:
`Path(top).glob("**/*")` give you all the files

auscompgeek, 2024-12-30:
`os.walk()` accepts path-like objects, so it shouldn't be too bad either way.

PeterJohnson, 2025-05-29:
Fixed by #7983.

closed 2025-05-29 by PeterJohnson