wpilibsuite/allwpilib #7615: Feature Request: Torrent Downloads

EmDash00, 2025-01-01:
It'd be nice if WPI could setup torrent downloads for WPILib to increase download speeds. It'd be as simple as setting up some dedicated torrent clients and then the community can leech and seed like any other torrent.

PeterJohnson, 2025-01-01:
Given that torrent downloads are blocked at most schools, and I'm not sure how downloads would be faster than what you can get from Cloudflare (where we host the files), and the logistics concerns of where we host seeding from, I'm not sure of the utility of this.

calcmogul, 2025-01-01:
I'm able to get 500 Mbps down from the CloudFlare hosting.

TheTripleV, 2025-01-01:
I peaked at [1100 Mbps](https://www.thesr71blackbird.com/Aircraft/Stories/sr-71-blackbird-speed-check-story) downloading the latest beta. If you'd like faster speeds on your existing internet plan, I would recommend using a multi threaded downloader.

closed 2025-01-01 by TheTripleV

KangarooKoala, 2025-01-01:
> I peaked at [1128 Mbps](https://www.thesr71blackbird.com/Aircraft/Stories/sr-71-blackbird-speed-check-story) downloading the latest beta.

What's this supposed to link to? Right now it links to a funny story.

sciencewhiz, 2025-01-01:
> > I peaked at [1128 Mbps](https://www.thesr71blackbird.com/Aircraft/Stories/sr-71-blackbird-speed-check-story) downloading the latest beta.
> 
> What's this supposed to link to? Right now it links to a funny story.

A funny story about comparing speeds, just like is happening in this thread

sciencewhiz, 2025-01-01:
@EmDash00 what download speeds are you getting? There have been scattered reports of slow download speeds in certain locations.

EmDash00, 2025-01-02:
@sciencewhiz

I think it was around 2 Mbps but I know I had faster speeds I bought at home. I can try a multithreaded downloader later.