wpilibsuite/allwpilib #7624: PR #7101 breaks AdvantageKit's `LoggedPowerDistribution`

tbowers7, 2025-01-03:
**Describe the bug**
Use of AdvantageKit's `LoggedPowerDistribution` relies on the symbol `wpi::raw_string_ostream::write_impl` that was removed with #7101 .

**To Reproduce**
Steps to reproduce the behavior:
1. Use AdvantageKit v4.0.0-beta-1 with WPILib release 2025.1.1
2. Attempt to deply and run on a RoboRIO 1.0

 - Link to code:

**Expected behavior**
Code runs without crashing.

**Screenshots**
Excerpt from the error log written on the Rio (`/tmp/hs_err_pid3300.log`):

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xad3dc1c8, pid=3300, tid=3303
#
# JRE version: OpenJDK Runtime Environment (17.0.9.7) (build 17.0.9.7-frc+0-2024-17.0.9u7-3)
# Java VM: OpenJDK Client VM (17.0.9.7-frc+0-2024-17.0.9u7-3, mixed mode, emulated-client, serial gc, linux-arm)
# Problematic frame:
# C  [libwpiutil.so+0x1321c8]  wpi::raw_string_ostream::write_impl(char const*, unsigned int)+0x10
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java
 again
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -XX:+UseSerialGC -Djava.lang.invoke.stringConcat=BC_SB -Djava.library.path=/usr/local/frc/third-party/lib -XX:+UnlockEx
perimentalVMOptions -XX:GCTimeRatio=5 -XX:+UseSerialGC -XX:MaxGCPauseMillis=50 /home/lvuser/Az-RBSI.jar

Host: rev 0 (v7l), 2 cores, 244M, NI Linux Real-Time - Academic 8.15
Time: Thu Jan  2 16:24:01 2025 MST elapsed time: 8.332697 seconds (0d 0h 0m 8s)

---------------  T H R E A D  ---------------

Current thread (0xb6413bd8):  JavaThread "main" [_thread_in_native, id=3303, stack(0xb657a000,0xb65ca000)]

Stack: [0xb657a000,0xb65ca000],  sp=0xb65c8768,  free space=313k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libwpiutil.so+0x1321c8]  wpi::raw_string_ostream::write_impl(char const*, unsigned int)+0x10

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  org.littletonrobotics.conduit.ConduitJni.configurePowerDistribution(II)V+0
j  org.littletonrobotics.conduit.ConduitApi.configurePowerDistribution(II)V+2
j  org.littletonrobotics.junction.inputs.LoggedPowerDistribution.<init>()V+25
j  org.littletonrobotics.junction.inputs.LoggedPowerDistribution.getInstance()Lorg/littletonrobotics/junction/inputs/LoggedPowerDistr
ibution;+10
j  frc.robot.util.PowerMonitoring.<init>(Lfrc/robot/util/LoggedTunableNumber;[Lfrc/robot/util/RBSISubsystem;)V+5
j  frc.robot.RobotContainer.<init>()V+668
j  frc.robot.Robot.robotInit()V+236
j  org.littletonrobotics.junction.LoggedRobot.startCompetition()V+1
j  edu.wpi.first.wpilibj.RobotBase.runRobot(Ljava/util/function/Supplier;)V+259
j  edu.wpi.first.wpilibj.RobotBase.startRobot(Ljava/util/function/Supplier;)V+137
j  frc.robot.Main.main([Ljava/lang/String;)V+5
v  ~StubRoutines::call_stub

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x00000005
...
```

**Desktop (please complete the following information):**
 - OS: [MacOS 14.7.1]
 - Project Information: WPILib Information:
Project Version: 2025.1.1
VS Code Version: 1.95.3
WPILib Extension Version: 2025.1.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.2024090204
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: /Users/tbowers/wpilib/2025/jdk
Vendor Libraries:
    AdvantageKit (4.0.0-beta-1)
    ChoreoLib (2025.0.0-beta-9)
    navx_frc (2025.1.1-beta-1)
    PathplannerLib (2025.0.0-beta-6.1)
    CTRE-Phoenix (v5) (5.34.0-beta-4)
    CTRE-Phoenix (v6) (25.0.0-beta-4)
    REVLib (2025.0.0-beta-4)
    ReduxLib (2025.0.0-beta2)
    URCL (2025.0.0-beta-1)
    WPILib-New-Commands (1.0.0)
    maplesim (0.2.7)
    photonlib (v2025.0.0-beta-8)
    YAGSL (2025.1.0.4)

**Additional context**
Running RIO image `FRC_roboRIO_2025_v2.0`

ThadHouse, 2025-01-03:
AdvantageKit's beta releases are only compatible with beta wpilib. We require all vendors to recompile for the in season release. Wait for them to do a season release.

closed 2025-01-03 by ThadHouse

tbowers7, 2025-01-03:
Thank you for the note about beta releases.

Can you comment on the stack trace that, to my eyes, seems to imply that the library `libwpiutil.so` is appearing to search for the function `write_impl()`, but even a mention of such a function in the WPILib codebase was removed with PR #7101 ?

```
Stack: [0xb657a000,0xb65ca000],  sp=0xb65c8768,  free space=313k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libwpiutil.so+0x1321c8]  wpi::raw_string_ostream::write_impl(char const*, unsigned int)+0x10
```

ThadHouse, 2025-01-03:
Not exactly. It's somehow trying to look for a code address that doesn't have an equivalent address most likely, and libwpiutil just happens to be the closest library loaded, so it assumes that. That is why the offset is actually fairly large.