wpilibsuite/allwpilib #7633: simulateJava fails on Linux Arm64

Daltz333, 2025-01-04:
**Describe the bug**
A blank 2025.1.1 Java project fails to simulateJava with the following error.

**To Reproduce**
1. Install WPILib 2025.1.1
2. Create a new Java project
3. Run `gradlew simulateJava` from command line

```
> Task :simulateJavaRelease FAILED
java.io.IOException: wpiutiljni could not be loaded from path.
        attempted to load for platform /linux/arm64/
Last Load Error:
/home/daltz/BlankJavaProject/build/jni/release/libwpiutiljni.so: libatomic.so.1: cannot open shared object file: No such file or directory
JVM Location: /usr/lib/jvm/java-17-openjdk-arm64/bin/java

        at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:47)
        at edu.wpi.first.util.WPIUtilJNI.<clinit>(WPIUtilJNI.java:44)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:470)
        at frc.robot.Main.main(Main.java:23)

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':simulateJavaRelease'.
> Process 'command '/usr/lib/jvm/java-17-openjdk-arm64/bin/java'' finished with non-zero exit value 1
```

**Expected behavior**
Functional simulation

**Desktop (please complete the following information):**
- Windows 11 Ubuntu 24.04 WSL ARM64 (WSL on Snapdragon X Elite)

Daltz333, 2025-01-04:
Also fails on JDK 21

Daltz333, 2025-01-04:
Workaround is `sudo apt install libatomic1`

ThadHouse, 2025-01-04:
We can document that, but because libatomic1 is the default on raspbian, which is what our arm builds actually target, we won't be able to "solve" this from our end.