wpilibsuite/allwpilib #7646: WPILIB Simulation Crashes on startup

SpaceStudios, 2025-01-06:
Whenever I start up a simulation on WPILIB, Glass correctly starts then it crashes giving the error EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ff98d8da420.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to WPILIB 2025.1.1 from WPILIB 2025.1.1 Beta 2

 - Link to code:
[Code](https://github.com/SpaceStudios/Maple-Sim-Kitbot-and-Kitbot--2025/)

PeterJohnson, 2025-01-06:
You’re mixing beta vendordeps (including akit) with release WPILib.  That’s not going to work, you need to upgrade your vendordeps to the release versions.

ThadHouse, 2025-01-06:
Addtionally, we see from that repo you manually upgraded the project from beta to release. That is not a supported scenario. You must reimport the projects between beta and release. In place upgrades from beta to release are not supported.

That is why the vs code extension vendor dep viewer is still showing beta vendor deps. Just updating the gradlerio version is not enough. Theres more that is required.

ThadHouse, 2025-01-15:
@SpaceStudios Do you still see this after updating everything to release? If not we should close this.

SpaceStudios, 2025-01-15:
I do not see this after updating to release

closed 2025-01-15 by SpaceStudios