wpilibsuite/allwpilib #7655: Severly outdated versions of library Skia Sharp make compiling on linux-based systems difficult.

ChloroticLeaf, 2025-01-08:
WPI Lib installer (2025) depends on version [80.2-81.0](https://github.com/mono/SkiaSharp/releases/tag/v2.80.1) of the [Skia Sharp](https://github.com/mono/SkiaSharp) library, which is 5 years out of date and contains an [8.8 severity CVE](https://nvd.nist.gov/vuln/detail/CVE-2023-4863).

This issue is making it extremely difficult to install WPI Lib on a Linux based system (Arch Linux, 6.12.3, x86_64). You can clone the release easily enough, but compiling the required Skia Sharp version requires running many non-editable scripts in a python2 venv. This makes it very annoying to compile.

A simple solution to this would be to update all code using Skia Sharp to a new version, or transition to a more common library like QT or GTK. This would allow new development with the graphics libraries to be implemented, (optimizations, bug fixes, etc.) and make the user experience much cleaner. It would also be nice for the installer to come with a get dependencies shell script, as this could potently solve many users' dependency problems.

ThadHouse, 2025-01-08:
The installer should be self contained, and contains a copy of skia sharp and all necessary libraries. What isn't working about that setup?

ChloroticLeaf, 2025-01-08:
This is the Error I get when running in a terminal emulator. 

Unhandled exception. System.TypeInitializationException: The type initializer for 'SkiaSharp.SKObject' threw an exception.
 ---> System.InvalidOperationException: The version of the native libSkiaSharp library (88.1) is incompatible with this version of SkiaSharp. Supported versions of the native libSkiaSharp library are in the range [80.2, 81.0).
   at SkiaSharp.SkiaSharpVersion.CheckNativeLibraryCompatible(Version minSupported, Version current, Boolean throwIfIncompatible)
   at SkiaSharp.SkiaSharpVersion.CheckNativeLibraryCompatible(Boolean throwIfIncompatible)
   at SkiaSharp.SKObject..cctor()
   --- End of inner exception stack trace ---
   at SkiaSharp.SKObject..ctor(IntPtr handle, Boolean owns)
   at SkiaSharp.GRGlInterface..ctor(IntPtr h, Boolean owns)
   at SkiaSharp.GRGlInterface.GetObject(IntPtr handle)
   at SkiaSharp.GRGlInterface.CreateOpenGl(GRGlGetProcedureAddressDelegate get)
   at Avalonia.Skia.GlSkiaGpu..ctor(IPlatformOpenGlInterface openGl, Nullable`1 maxResourceBytes) in /_/src/Skia/Avalonia.Skia/Gpu/OpenGl/GlSkiaGpu.cs:line 24
   at Avalonia.Skia.PlatformRenderInterface..ctor(ISkiaGpu skiaGpu, Nullable`1 maxResourceBytes) in /_/src/Skia/Avalonia.Skia/PlatformRenderInterface.cs:line 37
   at Avalonia.Skia.SkiaPlatform.Initialize(SkiaOptions options) in /_/src/Skia/Avalonia.Skia/SkiaPlatform.cs:line 20
   at Avalonia.SkiaApplicationExtensions.<>c__0`1.<UseSkia>b__0_0() in /_/src/Skia/Avalonia.Skia/SkiaApplicationExtensions.cs:line 20
   at Avalonia.Controls.AppBuilderBase`1.Setup() in /_/src/Avalonia.Controls/AppBuilderBase.cs:line 304
   at Avalonia.Controls.AppBuilderBase`1.SetupWithLifetime(IApplicationLifetime lifetime) in /_/src/Avalonia.Controls/AppBuilderBase.cs:line 179
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 174
   at WPILibInstaller.Program.Main(String[] args) in /home/runner/work/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/Program.cs:line 11
Aborted (core dumped)

ThadHouse, 2025-01-08:
Your installer zip does have a libSkiaSharp.so in in right?

ChloroticLeaf, 2025-01-08:
No. It only contains the installer binary.

ThadHouse, 2025-01-08:
There should be like 7 or 8 files in the download. I'm downloading right now to verify. But you at least should have a resources and artifacts zip, in addition to some other so's. Which file did you download, and from where?

ChloroticLeaf, 2025-01-08:
I am redownloading at the current moment

calcmogul, 2025-01-08:
For what it's worth, I run Arch Linux and have run the 2025 installer. I didn't have to install any extra libraries.

ChloroticLeaf, 2025-01-08:
ok, that's really odd. Redownloading, I now have the rest of the files.

ThadHouse, 2025-01-08:
More to the point, we actually explicitly depend on skia 2.88. So somehow it expecting 2.80 is wrong somewhere.

https://github.com/wpilibsuite/WPILibInstaller-Avalonia/blob/main/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia.csproj#L58

ChloroticLeaf, 2025-01-08:
hmm

ChloroticLeaf, 2025-01-08:
ill close this issue then

closed 2025-01-08 by ChloroticLeaf