wpilibsuite/allwpilib #7666: Add hypotnuse method to Distance

MrTinker64, 2025-01-12:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to do distance.in(distanceUnit) followed by some math and then unit.of(resultOfMyMath)

**Describe the solution you'd like**
The same way there is .div and .plus methods for Distance I would like a .hypot(otherDistance) method. Using the plus method at Line 59 of Distance as an example I would write it like this:
```
@Override
default Distance hypot(Measure<? extends DistanceUnit> other) {
  return (Distance) unit().ofBaseUnits(Math.hypot(baseUnitMagnitude() + other.baseUnitMagnitude()));
}
```

**Additional context**
Here's what I'm currently using as a work around in my code:
```
public static Distance hypot(Distance a, Distance b) {
  DistanceUnit unit = Meters;
  return unit.of(Math.hypot(a.in(unit), b.in(unit)));
}
```

auscompgeek, 2025-01-12:
IMO this doesn't sound like it should be an instance method. Reading `a.hypot(b)` it's very unclear what operation it's meant to be performing on `a`. I'm not sure how one would explain/document this as an operation either.

On the other hand a static method `hypot(a, b)` would be significantly easier to explain.

MrTinker64, 2025-01-12:
So instead it would be `Distance.hypot(a, b)`? That sounds good to me!