wpilibsuite/allwpilib #7671: Mac Installer for 2025.2.1 doesn't work, doesn't have Start button

alinav, 2025-01-12:
**Describe the bug**
Mac Installer for 2025.2.1 doesn't work, doesn't have start button

**To Reproduce**
Steps to reproduce the behavior:
1. Download https://packages.wpilib.workers.dev/installer/v2025.2.1/macOSArm/WPILib_macOS-Arm64-2025.2.1.dmg
2. Open it
3. Double click on installer
4. Start button missing, instead its asking for "select Artifacts Files (~artifacts)
5. When you select the folder it just trying to get inside, it doesn't accept unzipped folder

 - Link to code:

**Expected behavior**
Based on documentation and on what 2025.1.1 was, we should have Start button with install options after

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: MacBook Sequoia 15.1
 - Project Information: [In Visual Studio Code, press the WPILib button and choose WPILib: Open Project Information. Press the copy button and paste the data here. If not using VS Code, please include WPILib version, Gradle version, Java version, C++ version (if applicable), and any third party libraries and versions]

**Additional context**
Add any other context about the problem here.

sciencewhiz, 2025-01-12:
Is the 2025.1.1 installer still mounted?

alinav, 2025-01-13:
Yes, still mounted

sciencewhiz, 2025-01-13:
eject it, and rerun the installer