wpilibsuite/allwpilib #7673: Problem when compiling with the latest version of fmt

jimwang118, 2025-01-13:
**Describe the bug**
Line   93: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Rotation2d.h(204): error C3615: constexpr function 'frc::Rotation2d::ToMatrix' cannot result in a constant expression
	Line  127: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Translation2d.h(97): error C3615: constexpr function 'frc::Translation2d::ToVector' cannot result in a constant expression
	Line  161: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(62): error C3615: constexpr function 'frc::Pose2d::Pose2d' cannot result in a constant expression
	Line  168: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(223): error C3615: constexpr function 'frc::Pose2d::ToMatrix' cannot result in a constant expression
	Line  202: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(58): error C3615: constexpr function 'frc::Transform2d::Transform2d' cannot result in a constant expression
	Line  209: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(97): error C3615: constexpr function 'frc::Transform2d::ToMatrix' cannot result in a constant expression
	Line  243: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(214): error C3615: constexpr function 'frc::PoseTo3dVector' cannot result in a constant expression
	Line  277: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(231): error C3615: constexpr function 'frc::PoseTo4dVector' cannot result in a constant expression
	Line  311: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(325): error C3615: constexpr function 'frc::PoseToVector' cannot result in a constant expression
	Line  348: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Rotation2d.h(204): error C3615: constexpr function 'frc::Rotation2d::ToMatrix' cannot result in a constant expression
	Line  382: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Translation2d.h(97): error C3615: constexpr function 'frc::Translation2d::ToVector' cannot result in a constant expression
	Line  416: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(62): error C3615: constexpr function 'frc::Pose2d::Pose2d' cannot result in a constant expression
	Line  423: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(223): error C3615: constexpr function 'frc::Pose2d::ToMatrix' cannot result in a constant expression
	Line  457: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(58): error C3615: constexpr function 'frc::Transform2d::Transform2d' cannot result in a constant expression
	Line  464: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(97): error C3615: constexpr function 'frc::Transform2d::ToMatrix' cannot result in a constant expression
	Line  498: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(214): error C3615: constexpr function 'frc::PoseTo3dVector' cannot result in a constant expression
	Line  532: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(231): error C3615: constexpr function 'frc::PoseTo4dVector' cannot result in a constant expression
	Line  566: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(325): error C3615: constexpr function 'frc::PoseToVector' cannot result in a constant expression
	Line  603: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Rotation2d.h(204): error C3615: constexpr function 'frc::Rotation2d::ToMatrix' cannot result in a constant expression
	Line  637: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Translation2d.h(97): error C3615: constexpr function 'frc::Translation2d::ToVector' cannot result in a constant expression
	Line  671: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(62): error C3615: constexpr function 'frc::Pose2d::Pose2d' cannot result in a constant expression
	Line  678: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(223): error C3615: constexpr function 'frc::Pose2d::ToMatrix' cannot result in a constant expression
	Line  712: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(58): error C3615: constexpr function 'frc::Transform2d::Transform2d' cannot result in a constant expression
	Line  719: E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(97): error C3615: constexpr function 'frc::Transform2d::ToMatrix' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(214): error C3615: constexpr function 'frc::PoseTo3dVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(231): error C3615: constexpr function 'frc::PoseTo4dVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(325): error C3615: constexpr function 'frc::PoseToVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Rotation2d.h(204): error C3615: constexpr function 'frc::Rotation2d::ToMatrix' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Translation2d.h(97): error C3615: constexpr function 'frc::Translation2d::ToVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(62): error C3615: constexpr function 'frc::Pose2d::Pose2d' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Pose2d.h(223): error C3615: constexpr function 'frc::Pose2d::ToMatrix' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(58): error C3615: constexpr function 'frc::Transform2d::Transform2d' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/geometry/Transform2d.h(97): error C3615: constexpr function 'frc::Transform2d::ToMatrix' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(214): error C3615: constexpr function 'frc::PoseTo3dVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(231): error C3615: constexpr function 'frc::PoseTo4dVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/StateSpaceUtil.h(325): error C3615: constexpr function 'frc::PoseToVector' cannot result in a constant expression
E:\vcpkg\buildtrees\wpilib\src\v2025.2.1-d890fcf0d0.clean\wpimath\src\main\native\include\frc/system/plant/LinearSystemId.h(48): error C3615: constexpr function 'frc::LinearSystemId::ElevatorSystem' cannot result in a constant expression
[... cut ...]
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - VS: 2022 17.11.5
 - Project Information: [In Visual Studio Code, press the WPILib button and choose WPILib: Open Project Information. Press the copy button and paste the data here. If not using VS Code, please include WPILib version, Gradle version, Java version, C++ version (if applicable), and any third party libraries and versions]

**Additional context**
Add any other context about the problem here.

calcmogul, 2025-01-13:
Those errors don't seem related to fmtlib. The functions it complains about don't even use fmtlib.

jimwang118, 2025-01-13:
> Those errors don't seem related to fmtlib. The functions it complains about don't even use fmtlib.

Yes, judging from the error, it has nothing to do with fmt. It should be an issue with constexpr adaptation.

PR #7676 (2025-01-13): [wpimath] Fix various constexpr support bugs
Fixes #7673.
closed 2025-01-13 by PeterJohnson