wpilibsuite/allwpilib #7679: [NTCore] Clarification and Safeguards for pub/subuid Constraints in NetworkTables

cjlawson02, 2025-01-14:
Hi team,

I noticed that for NetworkTables publishers, the `pubuid` and `subuid` is currently constrained to fit within a signed 32-bit integer space ([source](https://github.com/wpilibsuite/allwpilib/blob/a14545102f2c16f8b446e765af8d872e028a92f8/ntcore/src/main/native/cpp/net/Message.h#L26)). However, this requirement is not explicitly documented in the relevant sections on the [networktables4.adoc](https://github.com/wpilibsuite/allwpilib/blob/main/ntcore/doc/networktables4.adoc#publish-request-message-publish) file.

This limitation became apparent when my UUID generator produced values exceeding the signed 32-bit positive integer range. As a result, the `pubuid` returned to the client was different than the value sent. Specifically:
- A `pubuid` of `2147483646` (just below the signed 32-bit overflow limit) worked as expected.
- A `pubuid` of `2147483647` caused the robot code to crash. Interestingly, not all values above the overflow limit cause a crash, but values  right above the boundary exhibit this behavior.

### Potential Security Concern
This behavior could pose a vulnerability. If a malicious actor gained websocket access, they could potentially send a message like the following:
```json
[{"method":"publish","params":{"type":"string","name":"/MyTable/myvalue","pubuid":2147483647,"properties":{}}}]
```
which would lead to potential denial-of-service scenarios.

### Suggested Solutions
Some solutions could be:
1. Changing the `pubuid` and `subuid` to be an **unsigned integer** with server-side validation.
2. Increasing the size to **unsigned 64-bit integers** with server-side validation.
3. Allowing the `pubuid` and `subuid` to support **strings** for greater flexibility.

Additionally, I request clarification and updates to the documentation to specify the current constraint for `pubuid` and `subuid` values, helping prevent similar issues for other developers and mitigating potential vulnerabilities.

Thanks for considering this! Please let me know if further details or tests are needed.

PeterJohnson, 2025-01-14:
I will document and add implementation validation of the current signed 32-bit values.  At some point (e.g. for 2027, when the platform goes to 64-bit), I may add support for *signed* 64-bit values.  I do not plan on changing the value type to strings or unsigned integers.

PeterJohnson, 2025-01-16:
Investigating further, the 32-bit range won't support the *full* 32-bit range, because (as you discovered) some of these values are used for sentinel values in the internal implementation.  So the spec will likely say something like the server "should" support signed 32-bit values, but may be constrained to smaller ranges based on implementation.  It will also recommend that clients use as small of values as possible for pubuid's, because those IDs are used in the MessagePack messages from the client to server (so larger values mean they take more space and time to encode/decode for every message).

cjlawson02, 2025-01-18:
Does the client need to worry about a collision with another pubuid, or could it just increment from 0?

PeterJohnson, 2025-01-20:
pubuid's are client-unique; there is no possibility of colliding with another client's pubuids.  So it's perfectly fine to just increment from 0.

PR #7726 (2025-06-19): [ntcore] Check id ranges in control messages
Fixes #7679.
closed 2025-06-19 by PeterJohnson