wpilibsuite/allwpilib #7680: [ntcore] Server requires subscription for announce, no response on publish

cjlawson02, 2025-01-14:
**Describe the bug**  
When a client publishes a topic, it does not receive an `announce` response from the server unless the client is also subscribed to the same topic. According to the [protocol spec](https://github.com/wpilibsuite/allwpilib/blob/main/ntcore/doc/networktables4.adoc#publish-request-message-publish), the server should send an `announce` in response to a client's `publish` message. Additionally, WebSocket pings and pongs are not exchanged until a subscription is made.

**To Reproduce**  
Steps to reproduce the behavior:
1. Publish a topic to the server without subscribing to it:  
   ```json
   [{"method":"publish","params":{"type":"string","name":"/MyTable/autoMode","pubuid":973718445,"properties":{}}}]
   ```
2. Observe that no `announce` response is received.
3. Then, send a subscription request for the same topic:  
   ```json
   [{"method":"subscribe","params":{"topics":["/MyTable/autoMode"],"subuid":25935943,"options":{}}}]
   ```
4. Notice that the server now sends an `announce` response containing the `pubuid`:  
   ```json
   [{"method":"announce","params":{"id":59,"name":"/MyTable/autoMode","properties":{},"pubuid":973718445,"type":"string"}}]
   ```
5. Observe that WebSocket pings and pongs also begin only after the subscription.

**Expected behavior**  
The client should receive an `announce` response from the server immediately after publishing a topic, regardless of whether it is subscribed to that topic. Additionally, WebSocket pings and pongs should occur as expected without requiring a subscription.

**Desktop (please complete the following information):**
- OS: macOS
- Project Information: Robot simulator on getting started example

**Additional context**  
The lack of an `announce` response for published topics without a subscription can cause issues for clients expecting confirmation or awareness of the topic they just published. The dependency on subscriptions for WebSocket ping/pong behavior might also indicate a related issue in connection handling. 

---

Wireshark dump for further debugging: [NTCoreWiresharkDump.zip](https://github.com/user-attachments/files/18405054/NTCoreWiresharkDump.zip)

PeterJohnson, 2025-01-14:
Thanks for the report.  I noticed the spec also needs updating to clarify in both the subscribe and announce message sections that announce is sent in response to a subscribe message for existing topics that have not yet been announced to the client.  This was an oversight when we changed the spec from announcing on connection to announcing on subscribe.

cjlawson02, 2025-01-14:
Does this mean that the client will receive announcements for topics it hasn't requested a subscription for? I was under the impression that new announcements are only sent when the subscription is created, when a topic starts publishing, or when a prefix topic has a new subtopic.

PeterJohnson, 2025-01-14:
No--your original impression is correct.

shueja, 2025-02-10:
This bug also takes the form of:
* Have a topic set as retained on the server (created but with no value)
* Subscribe to it before publishing
* Get an "announce" message
* Publish to it
* Get no "announce" message

DatAsianBoi123, 2025-04-15:
It also seems like no announce is sent if another publisher decides to publish to the same topic:
* publish to a topic (let's call it `/topic`)
* receive an `announce` response
* publish to `/topic` again
* no `announce` response

This behavior occurs regardless of if the first publisher is unpublished by the time the second publisher publishes or not.

cjlawson02, 2025-12-28:
@PeterJohnson Opened #8515 to fix this issue. It also fixes retained topics not receiving announcements when clients republish after reconnection.

closed 2026-08-17 by PeterJohnson