wpilibsuite/allwpilib #7700: DriverStation.isJoystickConnected Does not correctly report disconnected joysticks.

m10653, 2025-01-17:
**Describe the bug**
The methods `DriverStation.isJoystickConnected(0)` and `controller.isConnected()` produce incorrect results when controllers are reconnected. Starting with no controllers plugged in, the methods correctly report false. Plugging in a controller updates the status to true, as expected. However, disconnecting the controller does not update the status back to false.

Moving the controller to a different port in the Driver Station forces the status to update. Restarting the Driver Station does not resolve the issue or update the connection status.

**To Reproduce**
Steps to reproduce the behavior:
1. Have no controllers plugged in.
2. Start driver station
3. View state of `DriverStation.isJoystickConnected(0)`
4. Connect Controller to port 0
5. View state of `DriverStation.isJoystickConnected(0)` (Should now be true)
6. Disconnect controller
7. View state of `DriverStation.isJoystickConnected(0)` Value is still true)

 - Link to code:

**Expected behavior**
The connected state between the driver station and the robot code should be the same and correctly reflect the connected state of the Controller.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Project Information: WPILib Information:
Project Version: 2025.1.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.1.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.2024090204
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    AdvantageKit (4.0.0)
    PathplannerLib (2025.1.1)
    CTRE-Phoenix (v6) (25.1.0)
    ReduxLib (2025.0.0)
    REVLib (2025.0.0)
    Studica (2025.0.0)
    URCL (2025.0.0)
    WPILib-New-Commands (1.0.0)

**Additional context**
Using a xbox360 controller. Also verified same behavior on off brand controller as well.

Interesting note the `DriverStation.getJoystickType(0)` call does register a change on disconnect of a controller. It changes from 1 (when connected) to 255 (when disconnected)

sciencewhiz, 2025-01-17:
> **Additional context** Using a xbox360 controller. Also verified same behavior on off brand controller as well.

When you say an offbrand controller, is it also for xbox360? The driver station handles xbox360 controllers differently then other types of joysticks.

m10653, 2025-01-18:
Its not a xbox licensed controller. But follows the xbox (1?) form factor. Think it uses the xinput api ( would need to check) I verified I had the same issue with an official 360 controller, and I can test a Logitech joystick later on if needed. The work around I have so far is just checking `isconnected || getType == null`

sciencewhiz, 2025-01-21:
I think this might be a corner case. If you have two joysticks connected and then disconnect joystick 1, is it detected correctly?