wpilibsuite/allwpilib #7703: ADIS16470 can not be found Roborio 2.0

BCITAcosta, 2025-01-17:
While trying to use the Analog Devices ADIS16470 IMU on a RoboRio 2.0. It errors out in initalization with "ERROR 1 Could not find an ADIS16470". There was an issue use previously mentioned and I just wasn't sure if it was appropriate to bump it: https://github.com/wpilibsuite/2025Beta/issues/45. Upon inspecting the ADIS16470 class definition it seems that the Product ID was never changed, although the issue was closed. This is happening with the following release information:

 WPILIB Information:
Project  Version: 2025.1.1
VS Code Version: 1.96.2
WPILIB Extension Version: 2025.1.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version: 0.24.1
Java Version: 17
Vendor Libraries:
    REVLib (2025.0.0)
    WPILib-New-Commands (1.0.0)

Further background information: We did successfully use the ADIS16470 IMU with RoboRio 1.0 in 2024 and the WPILib 2024.1.1 release that has the same product ID in the class definition. We have tested it with our back up 16470 boards and they all work on the RoboRio 1.0 but not the 2.0 at this current moment.

PeterJohnson, 2025-01-17:
Yep, it looks like we fixed the register reading but not the actual product ID value.

BCITAcosta, 2025-01-17:
Yeah, the Roborio 1.0 has no issue initializing it with the 2025 Release but the 2.0 errors out, this is test with the same code between both Rios and same IMU board.

KangarooKoala, 2025-01-20:
Was this fixed by #7704 or is more action needed?

Braykoff, 2025-01-20:
I can confirm that our RoboRIO 2.0, running WPILib 2025.2.1, works fine with our ADIS16470 imu. It could be a physical issue with your RoboRIO, like something conductive shorting the pins of the SPI port.

closed 2025-01-22 by PeterJohnson