wpilibsuite/allwpilib #7717: Rumble Support for PlayStation Controller

zachfeelin, 2025-01-21:
I find WPIlib very useful for FRC, but something I'm finding really annoying is how there is no native rumble support for different PlayStation controllers without third party software and changing my team's code from being PlayStation controller based to being an Xbox controller based, with the added risk of our code breaking.

I would love it if there would be native support within the WPIlib libraries to use rumble with my PlayStation controller.

My team has considered using DS4Windows, but came to the conclusion that it would take too much effort to translate our code from being PlayStation controller based to Xbox controller based with the risk of something breaking or rumble not even working at all.

Adding support for this would be much appreciated, I know my team and others would greatly appreciate it! Thank you! :)

ThadHouse, 2025-01-21:
This is something that needs to be implemented on the DS side. Not something WPILib has direct control over.

sciencewhiz, 2025-01-21:
Is there native support for PS4/5 rumble in Windows?

closed 2025-02-02 by zachfeelin