wpilibsuite/allwpilib #7718: MacOS simulator crashes when Xbox controller is connected

mjansen4857, 2025-01-22:
**Describe the bug**
If an Xbox controller is connected to a macOS computer (in my case it's wireless through bluetooth) the simulator will crash when run. Disconnecting the controller allows the simulator to run normally.

Console output:
```
> Task :simulateJavaRelease FAILED
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
********** Robot program starting **********
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x0000000197137458, pid=33322, tid=259
#
# JRE version: OpenJDK Runtime Environment Temurin-17.0.12+7 (17.0.12+7) (build 17.0.12+7)
# Java VM: OpenJDK 64-Bit Server VM Temurin-17.0.12+7 (17.0.12+7, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, bsd-aarch64)
# Problematic frame:
# C  [CoreFoundation+0x14f458]  CF_IS_OBJC+0x18
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/mjansen/Dev/RobotCode2025/hs_err_pid33322.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.10.2/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.
7 actionable tasks: 6 executed, 1 up-to-date
NT: Listening on NT3 port 1735, NT4 port 5810

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':simulateJavaRelease'.
> Process 'command '/Users/mjansen/wpilib/2025/jdk/bin/java'' finished with non-zero exit value 134

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 1s
4:09:32 PM: Execution finished 'simulateJava'.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Xbox controller to macOS system
2. Launch robot code simulator

**Expected behavior**
Simulator should run when controllers are connected, and allow them to be mapped/used.

**Desktop (please complete the following information):**
 - OS: macOS 15.1.1
 - Project Information: 
 ```
WPILib Information:
Project Version: 2025.2.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.2.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: /Users/mjansen/wpilib/2025/jdk
Vendor Libraries:
    AdvantageKit (4.0.0)
    PathplannerLib (2025.2.1)
    CTRE-Phoenix (v6) (25.2.0)
    WPILib-New-Commands (1.0.0)
```

KangarooKoala, 2025-01-23:
I'll work on some more detailed testing when I have time (which might not be for a while), but I was able to run simulation on the :devRobot project in allwpilib with a Logitech controller connected.

sciencewhiz, 2025-02-01:
I can't reproduce this in the SimpleDifferentialDriveSimulation Example, although I only have a wired xbox 360 controller

Is this on Intel or Apple Silicon? Does it happen if AdvantageKit isn't used? I noticed that you have AdvantageKit 4.0.0 which isn't compatible with WPILib 2025.2.1

sciencewhiz, 2025-02-01:
I tried it with a wireless xbox controller also, and it also didn't crash, and I tried it with the AdvantageKit_DiffDriveTemplate as well. I'm on a apple silicon Mac mini with macOS 15.2/3

mjansen4857, 2025-02-02:
I'm on apple silicon with macOS 15.1.1. This happens even in a freshly created, empty robot project. Here's the output from that:
```
********** Robot program starting **********
NT: could not open persistent file 'networktables.json': No such file or directory (this can be ignored if you aren't expecting persistent values)
NT: Listening on NT3 port 1735, NT4 port 5810
********** Robot program startup complete **********
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x0000000197137458, pid=94224, tid=259
#
# JRE version: OpenJDK Runtime Environment Temurin-17.0.12+7 (17.0.12+7) (build 17.0.12+7)
# Java VM: OpenJDK 64-Bit Server VM Temurin-17.0.12+7 (17.0.12+7, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, bsd-aarch64)
# Problematic frame:
# C  [CoreFoundation+0x14f458]  CF_IS_OBJC+0x18
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/mjansen/Dev/controller_test/hs_err_pid94224.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
zsh: abort      /usr/bin/env    /Users/mjansen/wpilib/2025/jdk/bin/java   frc.robot.Main
```

I can try updating to 15.3 and see if that does anything. Also, I'm specifically using an elite controller, not sure if that makes a difference. I don't have any wireless normal controllers to try. The controller does work fine when wired.

mjansen4857, 2025-02-02:
Still occurs on 15.3

KangarooKoala, 2025-02-02:
> `# An error report file with more information is saved as:`
`# /Users/mjansen/Dev/controller_test/hs_err_pid94224.log`

Could you send `/Users/mjansen/Dev/controller_test/hs_err_pid94224.log`?

mjansen4857, 2025-02-02:
[hs_err_pid94224.log](https://github.com/user-attachments/files/18633800/hs_err_pid94224.log)

ThadHouse, 2025-02-02:
That crash very much looks like this fix in GLFW

https://github.com/glfw/glfw/commit/2c1d31009fdc527d5a44ff887659a25ecbf1f7f0#diff-d26bed262e1dde179ef7b80270b5ad9a3c13dcc37955cdd527248f0ce7f66d14

Which isn't really a fix, but more of just a won't crash. Our version of glfw is older than that.

sciencewhiz, 2025-02-03:
In privacy and security, do you see anything that's requested input monitoring or bluetooth permission? I don't on mine.

mjansen4857, 2025-02-03:
Yes, but if I disable them it still crashes.

crueter, 2025-03-01:
FTR, I also can't reproduce this when using a DS4 controller.

auscompgeek, 2026-11-02:
#9055 (released in 2027 alpha 7) switched the sim GUI's joystick handling to SDL3. I presume this is no longer an issue since then?