wpilibsuite/allwpilib #7722: Wpical output question

ShmayaR, 2025-01-23:
Hi! I’ve recently started using wpical to calibrate the April tags on my field. I’ve realized that in the outputted Json file there are a few lines of code that are preventing the simulation from running. This code is nearly identical to 2025 April tag field layout json file but at the end it has an extra pose with no ID. if I remove this code everything seems to run properly. does anyone know what this last pose is for?

the 2025 game field json which lets the simulation run properly:
```
{
  "tags": [
    {
      "ID": 1,
      "pose": {
        "translation": {
          "x": 16.697198,
          "y": 0.65532,
          "z": 1.4859
        },
        "rotation": {
          "quaternion": {
            "W": 0.4539904997395468,
            "X": 0.0,
            "Y": 0.0,
            "Z": 0.8910065241883678
          }
        }
      }
    },
    {
      "ID": 2,
      "pose": {
        "translation": {
          "x": 16.697198,
          "y": 7.3964799999999995,
          "z": 1.4859
        },
        "rotation": {
          "quaternion": {
            "W": -0.45399049973954675,
            "X": -0.0,
            "Y": 0.0,
            "Z": 0.8910065241883679
          }
        }
      }
    },
    {
      "ID": 3,
      "pose": {
        "translation": {
          "x": 11.560809999999998,
          "y": 8.05561,
          "z": 1.30175
        },
        "rotation": {
          "quaternion": {
            "W": -0.7071067811865475,
            "X": -0.0,
            "Y": 0.0,
            "Z": 0.7071067811865476
          }
        }
      }
    },
    {
      "ID": 4,
      "pose": {
        "translation": {
          "x": 9.276079999999999,
          "y": 6.137656,
          "z": 1.8679160000000001
        },
        "rotation": {
          "quaternion": {
[... 741 lines cut ...]
                }
            }
        },
        {
            "pose": {
                "rotation": {
                    "quaternion": {
                        "W": 0.35114692668777203,
                        "X": 0.8895263884556711,
                        "Y": -0.18386634973611068,
                        "Z": 0.2272263311182522
                    }
                },
                "translation": {
                    "x": 4.975261414530195,
                    "y": 10.107371606412972,
                    "z": -0.012881997722484717
                }
            }
        }
    ]
}

```
the extra pose is the one at the end.

ElliotScher, 2025-01-23:
Would you be able to share your camera calibration parameters?

Example:

[image]

ShmayaR, 2025-01-23:
MRcal
Square Width:1
Marker Width:0.75
Board Width:8
Board Height:8
Image Width:3840
Image Height: 2160

ElliotScher, 2025-01-24:
Can you verify that when you are calibrating, your field calibration directory only contains your field calibration video?

ElliotScher, 2025-01-24:
Make sure that you are using the same camera/lens for the camera calibration and the field calibration, other than that there a few more things to try:

If you could, try updating to WPILib version 2025.2.1, there were a couple of changes that occurred in that update and it may fix the issue.

If that doesn't work, please try printing out a ChArUco board with the 5x5 dictionary. (they can be generated here: https://calib.io/pages/camera-calibration-pattern-generator), as that is currently the only wpical compatible charuco dictionary (This is currently a feature being worked on: https://github.com/wpilibsuite/allwpilib/issues/7644)

ShmayaR, 2025-01-24:
ive solved the issue. in the video of the field wpical Identified a random object as an April tag for just a second but because it wasn't a real April tag it had no id and therefor wouldn't let the simulation run.
thank you, I wouldn't have found the problem if you hadn't tried to help.

ElliotScher, 2025-01-24:
Sounds good!

sciencewhiz, 2025-01-24:
Seems like there's still a bug that the tool output a tag without an id.

ElliotScher, 2025-01-24:
I think it would be ok adding a check to determine if all the JSON information is present for each tag. That way if something like this happens again, it removes the entire block.