wpilibsuite/allwpilib #7759: SwerveDriveKinematics toTwist2d method not returning expected twist given module deltas

viggy96, 2025-02-03:
**Describe the bug**
The `toTwist2d` method in SwerveDriveKinematics is not returning the expected resulting twist value.

**To Reproduce**
Steps to reproduce the behavior:
1. Run project
2. Debug

 - Link to code: https://github.com/lasarobotics/PurpleSwerve

**Expected behavior**
A non-zero twist should be returned given module deltas of around 0.1m for all 4 modules.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Project Information: 
```
WPILib Information:
Project Version: 2025.2.1
VS Code Version: 1.95.1
WPILib Extension Version: 2025.1.1
C++ Extension Version: Not Installed
Java Extension Version: 1.39.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: /home/vignesh/wpilib/2025/jdk
Vendor Libraries:
    AdvantageKit (4.1.0)
    PathplannerLib (2025.2.2)
    CTRE-Phoenix (v5) (5.35.1)
    CTRE-Phoenix (v6) (25.2.1)
    REVLib (2025.0.2)
    ReduxLib (2025.0.1)
    Studica (2025.0.1)
    ThriftyLib (2025.0.2)
    URCL (2025.0.1)
    WPILib-New-Commands (1.0.0)
    photonlib (v2025.1.1)
```

**Additional context**
You can see all the module delta values in the screenshot, and the resulting twist should NOT be zero for dX, dY, and dTheta.

viggy96, 2025-02-03:
The module coordinates are as follows:
```
(0.2794, 0.2794)
(0.2794, -0.2794)
(-0.2794, 0.2794)
(-0.2794, -0.2794)
```

KangarooKoala, 2025-02-03:
Double check that the module order is consistent (and that the inversions are correct). From looking at https://www.desmos.com/calculator/qiyoh688ht, it looks like either your FL and BR are swapped or inverted, or your FR and BL are swapped or inverted, depending on what the actual direction of rotation is. AdvantageScope has a swerve tab that is very good for getting the same kind of view of the module translations in real time.
To tell if the modules are swapped vs inverted, try translation. If translation works, then inversion is not an issue and it must be module swapping.

viggy96, 2025-02-03:
Looking through my code further, it appears I forgot to add module offsets when in simulation.

closed 2025-02-03 by viggy96