wpilibsuite/allwpilib #7760: [wpimath] ExponentialProfile can produce oscillating setpoints

jwbonner, 2025-02-03:
Under certain conditions, an exponential profile with a constant goal can produce oscillations that prevent it from stopping at the goal. The graph below shows the position setpoint for one such profile.

[image]

This behavior appears to be extremely sensitive to the initial inputs. The example below produces the issue consistently, but changing the goal velocity to `2.99999` or `3.00001` fixes the issue. It's not clear whether this might be related to using a nonzero goal velocity given how sensitive it is to the exact values.

```java
ExponentialProfile profile = new ExponentialProfile(
    ExponentialProfile.Constraints.fromStateSpace(206.18996798292426, -4.0508214770151705, 0.07795421563066239));
ExponentialProfile.State state = new ExponentialProfile.State(0.1, 0.0);
ExponentialProfile.State goal = new ExponentialProfile.State(1.75, 3);

@Override
public void robotPeriodic() {
  SmartDashboard.putNumber("Position", state.position);
  SmartDashboard.putNumber("Velocity", state.velocity);
}

@Override
public void teleopPeriodic() {
  state = profile.calculate(0.02, state, goal);
}
```

This log includes the positions and velocities of the setpoints for the example above: [exponentialprofilebug.wpilog.zip](https://github.com/user-attachments/files/18645763/exponentialprofilebug.wpilog.zip)

**Desktop (please complete the following information):**
 - OS: macOS 15.2
 - Project Information:

```
WPILib Information:
Project Version: 2025.2.1
VS Code Version: 1.96.4
WPILib Extension Version: 2025.2.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: /Users/jonah/wpilib/2025/jdk
Vendor Libraries:
```

calcmogul, 2025-02-03:
This is intended behavior. If you provide a constant goal with a nonzero velocity, the system will enter a limit cycle to hit the goal repeatedly. If you want the profile to reach steady-state, you should pass a zero velocity.

jwbonner, 2025-02-03:
Why is the behavior entirely different with a velocity of 2.99999 and 3.00000 in that case? It seems like the resulting profile should be nearly identical in those cases.

jwbonner, 2025-02-03:
After the initial movement, the profile also fails to reach the goal again afterwards (and doesn’t follow a normal cycle). If it cycles to hit the goal repeatedly, I would have expected both of those to be true.

calcmogul, 2025-02-03:
Could you be more specific about the behavior you saw instead of a limit cycle?

calcmogul, 2025-02-03:
A position vs velocity plot would be much more informative than a time domain plot.

jwbonner, 2025-02-03:
This is with a setpoint of 3.00000:

[image]

And with a setpoint of 2.99999:

[image]

I marked the goal with a red X on both plots.

jwbonner, 2025-02-03:
(The first example is stable after it reaches the goal)

calcmogul, 2025-02-03:
It shouldn't be stable tho, because it's not physically possible to stay at a given position-velocity pair in steady-state.