wpilibsuite/allwpilib #7762: Conditional breakpoint evalutation fails -- where do I look for the problem?

chauser, 2025-02-03:
**Describe the question you have.**
When simulating one of my 2025 java robot projects with a conditional breakpoint set, a popup appears when the breakpoint location is reached with the message: 

Breakpoint condition '(this.m_positionCorrection > 0.01)' error: Cannot evaluate, please specify projectName in launch.json. 

Conditional breakpoints work in other 2025 java projects. I can't figure out what's different about the one where they don't work. The pointer to `launch.json` seems to be a red herring as `projectName` does not appear to be a valid field in the configurations listed there, according to Intellisense. Further, the contents of `launch.json` in this project are the same as in the projects where it works:
```
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [

    {
      "type": "wpilib",
      "name": "WPILib Desktop Debug",
      "request": "launch",
      "desktop": true,
    },
    {
      "type": "wpilib",
      "name": "WPILib roboRIO Debug",
      "request": "launch",
      "desktop": false,
    }
  ]
}
```
Where do I look next? Thanks.

ThadHouse, 2025-02-03:
You don't happen to have a space in the project name, right? Otherwise, thats not really anything we've seen.

You're right that error isn't great. Due to the native library requirements, we actually inject a custom launch.json into VS Code when you run either that command or the sim command.

If you run the `WPILib: Show Log Folder` command, that will open up the logs, which I _think_ has the launch.json contents we inject. You theoretically can replace the one in the project with that and it should work. Hasn't been tested in a while though. But you can also compare that to one of the projects that work.
]

chauser, 2025-02-04:
Here's the content of the log file. I added spaces between the lines for readability. The first pair, referencing robot-2025-practice, corresponds to a project where conditional breakpoints work. The second pair, referencing robot-2025, is for the project where they don't work. I'm not spotting any fundamental difference. However, on my windows machine the breakpoints work in the robot-2025 project so I'll move on to doing the work I need to do there and come back to this at a less critical time. Thanks, Thad, for the help.

```
2025-02-03T21:15:59.535Z verbose: executing command in workspace {"0":{"f":[],"c":"./gradlew simulateExternalJavaRelease -Dorg.gradle.java.home=\"/home/hauser/wpilib/2025/jdk\"","g":{"cwd":"/home/hauser/sciborgs-2025/robot-2025-practice","env":{"JAVA_HOME":"/home/hauser/wpilib/2025/jdk"}}},"1":"/home/hauser/sciborgs-2025/robot-2025-practice"}

2025-02-03T21:16:06.160Z verbose: Java Simulation:  {"0":{"console":"integratedTerminal","cwd":"/home/hauser/sciborgs-2025/robot-2025-practice","env":{"HALSIM_EXTENSIONS":"/home/hauser/sciborgs-2025/robot-2025-practice/build/jni/release/libhalsim_gui.so:","DYLD_LIBRARY_PATH":"/home/hauser/sciborgs-2025/robot-2025-practice/build/jni/release","LD_LIBRARY_PATH":"/home/hauser/sciborgs-2025/robot-2025-practice/build/jni/release"},"mainClass":"frc.robot.Main","name":"WPILib Java Simulate","request":"launch","stopOnEntry":false,"type":"java","vmArgs":"-Djava.library.path=\"/home/hauser/sciborgs-2025/robot-2025-practice/build/jni/release\""}}

2025-02-03T21:52:15.261Z verbose: executing command in workspace {"0":{"f":[],"c":"./gradlew simulateExternalJavaRelease -Dorg.gradle.java.home=\"/home/hauser/wpilib/2025/jdk\"","g":{"cwd":"/home/hauser/sciborgs-2025/robot-2025","env":{"JAVA_HOME":"/home/hauser/wpilib/2025/jdk"}}},"1":"/home/hauser/sciborgs-2025/robot-2025"}

2025-02-03T21:52:19.737Z verbose: Java Simulation:  {"0":{"console":"integratedTerminal","cwd":"/home/hauser/sciborgs-2025/robot-2025","env":{"HALSIM_EXTENSIONS":"/home/hauser/sciborgs-2025/robot-2025/build/jni/release/libhalsim_gui.so:","DYLD_LIBRARY_PATH":"/home/hauser/sciborgs-2025/robot-2025/build/jni/release","LD_LIBRARY_PATH":"/home/hauser/sciborgs-2025/robot-2025/build/jni/release"},"mainClass":"frc.robot.Main","name":"WPILib Java Simulate","request":"launch","stopOnEntry":false,"type":"java","vmArgs":"-Djava.library.path=\"/home/hauser/sciborgs-2025/robot-2025/build/jni/release\""}}
```

chauser, 2025-02-06:
Further experimentation pointed to state held outside of the project directory, since conditional breakpoints worked  when the same project was cloned at a new location on the same machine and for the same user. That observation eventually led me to `~/wpilib/2025/vscode/VSCode-linux-x64/data/user-data`. Moving that aside and letting it be regenerated got me a complaint about not having a jdk set up and an opportunity to fix the problem when the conditional breakpoint was hit. I selected wpilib's jdk and then things worked. I'm sure I lost settings choices and history such as the theme choice and recent projects.
I'm wasn't entirely convinced that everything would be set up correctly so I deleted the entire wpilib/2025 folder and re-installed. Things seem fine now and if someone else encounters the problem that would be my suggestion.

closed 2025-02-06 by chauser