wpilibsuite/allwpilib #7794: Make SysId's default step voltage smaller

calcmogul, 2025-02-16:
Motors are more powerful these days, which means the default voltage tends to make wheels slip, invalidating the test data.

TLNing260310, 2026-08-31:
Hi! I'd like to work on this, but I want to confirm the intended scope before opening a PR. #7795 changed the core Config default from 7 V to 4 V, while the current issue says the change should be made in the SysIdRoutine examples and is labeled component: examples. The earlier discussion also suggested starting at 2 V and increasing it when needed. Should the fix leave the library default at 7 V and explicitly configure 2 V (or 4 V) in the Java, C++, and Python examples? Should the corresponding wpilib-docs guidance be updated as well? I'm happy to make the matching changes once the value and scope are confirmed.

calcmogul, 2026-09-01:
I think we should do both. 7 V is almost always too much, but users should also be picking a good value specific to their system. If the docs talk about it, then yes, they should also be updated.