wpilibsuite/allwpilib #7804: Match number in logs is incorrect

prensing, 2025-02-19:
**Describe the bug**
Based on the results from Week 0, it looks like the match numbers recorded in the WPILib log files is 1 less than it should be. We have a log labeled as "Match 12" but that was really Match 13.   If this is an FMS bug, apologies, and let me know how to file it.

**Additional context**
I have not had a chance to check the DS logs.

prensing, 2025-02-19:
All the matches seem to be 1 less. The log files were renamed to the match number recorded inside the file.

We were running WPILib 2025.3.1.

ThadHouse, 2025-02-19:
Would definitely need to see what the DS logs show. We just record exactly what the DS gives us.

ThadHouse, 2025-02-19:
Also, is the match number logged through the match continuous? Or does it bump up by 1 at some point?

prensing, 2025-02-19:
I will get the DS log files tomorrow (forgot to grab them).
The Match number (in AdvantageScope:  "NT/FMSInfo/MatchNumber") jumps just once from 0 to a constant value.

Kevin-OConnor, 2025-02-19:
Amanda was able to review the logs from 190 and the DS logs also show Match - 1. I'll let the FMS folks know and we'll see if we can track it down before Week 1.

Kevin-OConnor, 2025-02-20:
Was able to reproduce and confirm this morning on one our test setups at HQ. We identified and fixed the FMS bug causing it so it should be resolved for Week 1 events. Thanks for bringing this to our attention!!

closed 2025-02-20 by Kevin-OConnor