wpilibsuite/allwpilib #7821: Use records in Java for classes with immutable data

Gold856, 2025-02-25:
## Rationale
[Java records](https://docs.oracle.com/en/java/javase/17/language/records.html) are a special type of class that allow you to create objects that contain immutable data in a much more terse way than with normal classes. The private fields, accessors, constructor, `equals`, `hashcode`, and `toString` are automatically implemented, with the accessors having the same name as the field. This means that we can drop the `get` prefix on several different accessors. If the behavior of any of the autoimplemented methods isn't desirable, they can be implemented manually to get the desired behavior. Even though we would end up manually implementing a few methods like `equals` or `toString` (the former is because of floating point, the latter is because of formatting floats,) records would still make the code for types like the geometry types less verbose.

In addition, you can use [record patterns](https://docs.oracle.com/en/java/javase/23/language/record-patterns.html#GUID-7623D3AD-4141-4914-A384-60C65BD0C010) to essentially destructure data, although that's bordering on misuse of the feature. It seems Java might add destructuring support using records without the need for instanceof/switch, and it would be good to be prepared for such a future (if that future ever comes).

## Goal
The idea would be to move most wpimath classes, specifically ones with immutable members, but others can be added if desired, to records. Some classes have accessors for non-fields, and we would drop the `get` prefix on those to maintain consistency. This would cause massive API breakage, and as such, is targeted for 2027.

PeterJohnson, 2025-02-25:
What classes appear to be good candidates for this?  Many of the math classes have significant custom functionality and as such I’m not sure how good of a fit records are.

Gold856, 2025-02-25:
The pose, rotation, transform, and translation classes. They all have several immutable member variables and have substantial boilerplate that could be removed for accessing all the members, constructing the object, `hashCode`, and `equals` (sometimes).

spacey-sooty, 2025-02-26:
> What classes appear to be good candidates for this? Many of the math classes have significant custom functionality and as such I’m not sure how good of a fit records are.

You can still apply a lot of custom functionality to records, but as Gold said it cuts down on boilerplate.

crueter, 2025-03-19:
> and we would drop the get prefix on those to maintain consistency

Would this also entail moving around C++ method names and removing `Get` prefixes? Dropping the get and just using `fieldName()` to access `m_fieldName` is pretty standard practice in C++, as is using camel case rather than title case.

calcmogul, 2025-03-19:
Snake case is also rather popular in C++ libraries these days.

PeterJohnson, 2025-03-20:
Will entry-level (think middle school level) programmers understand how Java records work?  Will they understand `getX()` better than `x()`?  Will they understand why it's `x()` and not just `x`?

If we're extending this more generally, `getX()` and `setX(value)` are very common idiomatically in Java; there are a few standard library classes that use just `x()` and `x(value)` but the extra `get` and `set` prefix probably makes usage easier to understand for brand-new programmers?

crueter, 2025-03-20:
> Will entry-level (think middle school level) students understand how Java records work? Will they understand `x()` better than `getX()`? Will they understand why it's `x()` and not just `x`?
> 
> If we're extending this more generally, `getX()` and `setX(value)` are very common idiomatically in Java; there are a few standard library classes that use just `x()` and `x(value)` but the extra `get` and `set` prefix probably makes usage easier to understand for brand-new programmers?

In reality, the most intuitive solution would be to use a public member variable, a la `x = ...` and access it via `x`. This of course doesn't let you do anything custom in your setters.

The ideal solution then would be allowing that syntax but still with custom setter/getter functionality. The best example is Qt (C++) which lets you define a "property" as such:

```cpp
private:
    int m_value = 0;
    Q_PROPERTY(value int READ value WRITE setValue)

public:
    int value() const { return m_value; }
    void setValue(const int newValue) { m_value = newValue; } // and any custom stuff can be done here
```

And then in QML, which can access C++ containers:

```qml
customClass.value = 10
console.log(customClass.value)
```

Of course this isn't really easy to implement. I know there's a property/MOC library in C++ but it still doesn't give you that special syntax afaict, let alone anything in Java. I suppose in C++ you could do some weird stuff with `operator int() const` etc. and `operator=()` but that would be incredibly cumbersome and not worth the effort.

In my experience teaching brand-new programmers, most seem to pretty equally understand both `getX()` and `x()`. It's not really intuitive when learning OOP anyway, because like I said, either way they're expecting it to be `class.x = ...` and `class.x`. If anything they seem to expect to NOT use the get prefix.

Should probably create a separate issue or discussion for this to be honest.

PeterJohnson, 2025-03-20:
Created a discussion here: https://github.com/wpilibsuite/allwpilib/discussions/7874

PeterJohnson, 2025-03-20:
I think we need to discuss whether records make sense in general for user-facing classes.  Separate from the getX/x naming difference to all other existing classes, the main concern is there's a lot of hidden behavior in records (generation of functions etc).  While it shortens the implementation, it's also less clear to someone who doesn't know Java well to understand what's going on--e.g. "I'm calling x(), where is that defined?"  What happens when you click on `x()` in user code to go to the definition?  Where does it take you to?

crueter, 2025-03-21:
> What happens when you click on x() in user code to go to the definition? Where does it take you to?

By default it goes to the definition in the record signature (i.e. clicking `<Rectangle>.length()` goes to the `record Rectangle(double length, double width)` signature)

If any methods are redefined, it properly goes to the reassigned method.

PeterJohnson, 2025-03-26:
What do the generated javadocs look like for records?  Does it expand out all the generated functions with generated docs?

crueter, 2025-03-26:
> What do the generated javadocs look like for records? Does it expand out all the generated functions with generated docs?

It doesn't appear to expand out the generated functions but rather documents the parameters based on the root javadoc. I found this example from someone.

[image]

PeterJohnson, 2025-03-26:
That's not super great for discoverability.  As a student looking at that, how do I know I need to call `name()` to get the name, instead of just `name`?  Intellisense helps, but still...

crueter, 2025-03-26:
> That's not super great for discoverability. As a student looking at that, how do I know I need to call `name()` to get the name, instead of just `name`? Intellisense helps, but still...

After a certain point you're likely better off using a public final field. No reason to worry about generating getter functions if you're not processing any data between the field and the output, though names would likely need to change to use TitleCase which is generally the standard for non-struct public fields.

PeterJohnson, 2025-03-26:
We don't use TitleCase for variable names anywhere in our codebase, including public final fields, and I wouldn't want to change to that.

PeterJohnson, 2025-04-01:
With some further discussion, particularly around the quality of Javadocs, we're planning to only use records for pure-data use cases such as simple pairs/tuples, and not for immutable data classes more broadly (e.g. where there's a lot of additional functionality).

closed 2025-04-22 by PeterJohnson