wpilibsuite/allwpilib #7822: Command Groups Final Methods

MBorchert, 2025-02-25:
Why would the command groups make their inherited methods final?

Inheriting from command groups is useful, and one section of the docs even recommends doing it. Why, then, would all command groups disallow this?

If they were not final, then isFinished could be utilized for end conditions which apply to the group, but not its members, code run for every execute, or custom logging be added. To access such features in our codebase, we've had to decompile each command, copy it into a 'NotFinalCommandGroup' and utilize that instead, which feels rather ridiculous. Why have they been made final?

PeterJohnson, 2025-02-25:
There's no need for decompilation, all the source code is available: https://github.com/wpilibsuite/allwpilib/tree/main/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command

The reason these particular functions are non-overridable is because overriding them is a footgun: https://github.com/wpilibsuite/allwpilib/issues/3536

MBorchert, 2025-02-25:
So the intended method for obtaining desired functionality *is* to copy code from github and remove 'final' keywords?

ArchdukeTim, 2025-02-25:
> and one section of the docs even recommends doing it. 

Are you referring to https://docs.wpilib.org/en/stable/docs/software/commandbased/organizing-command-based.html#subclassing-command-groups

Subclassing this way is really just aliasing, but doesn't actually extend the functionality.

> Inheriting from command groups is useful

Can you give an example of this?

MBorchert, 2025-02-26:
Yes, I was referring to that section of the docs.

Overriding isfinished would be a very useful way to make a command group which can end itself at any point, rather than forcing isfinisheds into commands which do't really want it or doing an awkward cancel. Overriding initialize is obviously useful for calculations occurring at the start of a command rather than when it is created. Overriding execute would be nice for custom logging, such as showing advantagekit which commands in the group are currently running, since there is currently no way to do that.

MBorchert, 2025-02-27:
Those are good points. I typically prefer a style based on extension, but I suppose the factory style could also be interesting. The specific scenario which prompted this issue was that I wanted to pause a commandgroup during execution, then resume it at precisely the same point later. Would you have any idea how to accomplish that?

KangarooKoala, 2025-02-27:
If the pausing can happen at any time (i.e., in the middle of a subcommand), I can't think of a clean way to do it. However, if you only care about pausing in between subcommands, you can add commands that will wait for the pause:
```java
public static Command pausableSequence(BooleanSupplier pause, Command... commands) {
  if (commands.length == 0) {
    return Commands.none();
  }
  BooleanSupplier canRun = () -> !pause.getAsBoolean();
  Command[] newCommands = new Command[2 * commands.length - 1];
  for (int i = 0; i < commands.length; ++i) {
    newCommands[2 * i] = command;
    if (i < commands.length - 1) {
      newCommands[2 * i + 1] = Commands.waitUntil(canRun);
    }
  }
  return Commands.sequence(newCommands);
}
```
Note that you must make a new wait command via `Commands.waitUntil(canRun)` every time, since commands can't be used multiple times in a composition (or in the same composition).

If you really do want to interrupt your commands in the middle of execution, a hacky way to do it is in the following:
> [!WARNING]
> It's easy for commands to assume one way or another that `execute()` will be called continuously until the command finishes, especially without any other commands running on the same subsystem in between. Violating this assumption can cause potentially subtle problems and should only be done at your own risk.
```java
public static Command addPause(BooleanSupplier pause, Command command) {
  return new WrapperCommand(command) {
    @Override
    public void execute() {
      if (!pause.getAsBoolean()) {
        super.execute();
      }
    }
  }
}

public static Command pausableSequence(BooleanSupplier pause, Command... commands) {
  // In case the user passed in an array, make a new array to avoid modifying that array
  Command[] newCommands = new Command[commands.length];
  for (int i = 0; i < commands.length; ++i) {
    newCommands[i] = addPause(pause, commands[i]);
  }
  return Commands.sequence(newCommands);
}
```

KangarooKoala, 2025-02-28:
Glad to have been able to help clear things up!