wpilibsuite/allwpilib #7830: WPILIB jars not included in tools

R1shy, 2025-02-27:
**Is your feature request related to a problem? Please describe.**
When I use the WPILib installer and pick to install tools only it doesn't download the jar files for WPILib.

**Describe the solution you'd like**
add the jar files for WPILib and it's dependencies to the list of things downloaded when installing tools only.

**Describe alternatives you've considered**
I got them anyway by using the "everything" option.

**Additional context**
I don't want to install the everything option because I am using a different editor (neovim) than the one provided

PeterJohnson, 2025-02-27:
The tools only option does exactly what it says on the tin, installs only the tools, nothing else.  Install everything is the only other available option, but if you select it, you can choose to skip install of vscode if you're using a different editor.

R1shy, 2025-02-27:
ok I didn't see the option to not install VSCode

closed 2025-02-27 by R1shy