wpilibsuite/allwpilib #7862: Server error on camera calibration export

anherm160, 2025-03-14:
**Describe the bug**
I'm getting a 500 error in when I try to export my camera calibration settings from the web client. I'm running the server (version 2025.2.1) on an OrangePi B, version. My camera is an Arducam_OV9871_USB. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect client computer to OrangePi's network
2. Open web client at http://photonvision.local:5800/#/cameras and run through camera calibration
3. Navigate to the specific calibration result and click "Export"
4. Observe the spawned web page showing a black background and the text "Server error". In my case the URL is http://photonvision.local:5800/api/utils/getCalibrationJSON?width=1280&height=720&cameraUniqueName=arducam_ov9281_usb_camera

**Expected behavior**
I expect to either see or be able to download a JSON file containing my camera's calibration settings.

**Screenshots**
This is the browser error I see:

[image]

I looked at journal logs on the Orange Pi and found this null pointer exception:

[image]

**Desktop (please complete the following information):**
I think this is irrelevant, but here it is anyway.
 - OS: Windows 11
 - Project Information: WPILib Information:
Project Version: 2025.3.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.3.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    libgrapplefrc (2025.0.8)
    maplesim (0.3.8)
    PathplannerLib (2025.2.5)
    CTRE-Phoenix (v5) (5.35.1)
    CTRE-Phoenix (v6) (25.3.1)
    photonlib (v2025.2.1)
    ReduxLib (2025.0.1)
    REVLib (2025.0.3)
    Studica (2025.0.1)
    ThriftyLib (2025.0.2)
    URCL (2025.0.1)
    WPILib-New-Commands (1.0.0)
    YAGSL (2025.7.1)

**Additional context**
Add any other context about the problem here.

sciencewhiz, 2025-03-14:
Is there a reason you think this is a WPILib bug instead of a PhotonVision bug?

spacey-sooty, 2025-03-16:
This looks to be a PhotonVision bug, if anyone has permissions to transfer the issue that would be good. If not it can be opened at https://github.com/PhotonVision/photonvision/issues

awordforthat, 2025-03-24:
my mistake - I'll report it there.

closed 2025-08-03 by auscompgeek