wpilibsuite/allwpilib #7868: Mechanism2d field not published to network table with smartdashboard

isaacngkh, 2025-03-19:
**Describe the bug**
We notice that we cannot see our mechanism2d in ascope. It seems the .controllable/.name/.type fields are empty after updating to 2025.3.2

**To Reproduce**
Steps to reproduce the behavior:
1. Publish mechanism2d with SmartDashboard.putData()
2. try to drag it into Poses in ascope in 3d field

 - Link to code:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Left is a log from today with 2025.3.2. RIght is a log from saturday with 2025.3.1
[image]

**Desktop (please complete the following information):**
 - OS: windows 11
 - Project Information: [In Visual Studio Code, press the WPILib button and choose WPILib: Open Project Information. Press the copy button and paste the data here. If not using VS Code, please include WPILib version, Gradle version, Java version, C++ version (if applicable), and any third party libraries and versions]
 
WPILib Information:
Project Version: 2025.3.2
VS Code Version: 1.96.2
WPILib Extension Version: 2025.3.2
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    DogLog (2025.4.0)
    PathplannerLib (2025.2.6)
    CTRE-Phoenix (v6) (25.3.1)
    photonlib (v2025.2.1)
    PlayingWithFusion (2025.01.23)
    REVLib (2025.0.3)
    WPILib-New-Commands (1.0.0)

**Additional context**
Add any other context about the problem here.

Gold856, 2025-03-19:
Unable to reproduce with a log from 2025.2.1. Are you sure that you a) let the data load in and b) actually are viewing the log at a timestamp where the Mechanism2d has been published? I can reproduce your issue if I scrub the log to a point before the Mechanism2d was published.
[image]

isaacngkh, 2025-03-19:
It's a problem we started seeing after updating to 2025.3.2 yesterday. We've not seen this problem before with older version of WPILib. Here is a [log ](https://drive.google.com/file/d/1z8cpqUCKLEJvI594HZ1GsaPXdC67Gunk/view?usp=sharing)file from yesterday after updating 2025.3.2

It is not just this one log file with this problem; it is every log file since updating 2025.3.2. We are using DogLog, so maybe it has something to do with DogLog instead of WPILib. Here is [our ](https://github.com/gwhs/2025-Reefscape/blob/dev/src/main/java/frc/robot/RobotVisualizer.java)code where we are publishing the Mechanism2d to SmartDashboard, and we log NT into log files with [DogLog](https://github.com/gwhs/2025-Reefscape/blob/dev/src/main/java/frc/robot/Robot.java#L39)

Gold856, 2025-03-19:
I examined the log using https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/examples/printlog/datalog.py and can confirm that the data for several of the other topics is missing. I haven't been able to root cause it though, and there's no changes to WPILib since 2025.3.1 that would affect that part of the code. It is possible you're seeing a weird interaction/race condition and 2025.3.2 just happened to trigger it. If you have time, could you start DogLog before RobotContainer is constructed? I'm curious to see if that will properly capture the data.

isaacngkh, 2025-03-20:
Now I am noticing that the [meta data](https://github.com/gwhs/2025-Reefscape/blob/dev/src/main/java/frc/robot/Robot.java#L41-L52) we are logging ourselves are missing from the log file, in additional to SystemStats that DogLog log for us. A [log ](https://drive.google.com/file/d/1iy7jK9JYl9tjoa1ewqQ-fBDH8NhXzXQj/view?usp=sharing)file from today. After moving dog log before robot container, and mechanism2d is properly showing up

Gold856, 2025-03-20:
Yeah, it sounds like you hit some sort of race condition where you started NT capture just after a bunch of data gets published and so the log misses that data. It's a little bit weird that data that gets published periodically seems to also be missing, but as long as it all gets resolved by turning on NT logging first, you're probably fine.

PeterJohnson, 2025-03-20:
Sounds like a bug in DogLog?  It should properly handle this case (starting it after some stuff has already been written to NT).

tssmith7, 2025-03-21:
I'm having the same problem and I don't use DogLog.  I will try to put together a test case but at this point it seems to work in Live Mode (NT) and the logs created using simulation but not the logs created on the RoboRio.

jonahsnider, 2025-03-30:
> Sounds like a bug in DogLog? It should properly handle this case (starting it after some stuff has already been written to NT).

Bit of a late reply but DogLog's NT capture is literally just a one line call to `DataLogManager`, so I'd be surprised if anything there is wrong/bugged:

https://github.com/jonahsnider/doglog/blob/c3cf42e1f620d4f1465ff8dee23f5f7258bdf23b/lib/src/main/java/dev/doglog/internal/log_thread/reporters/DataLogReporter.java#L183

isaacngkh, 2025-04-01:
Updated to doglog 2025.9.0 and the issue is fixed

closed 2025-04-01 by isaacngkh