wpilibsuite/allwpilib #7872: Make Color and Color8Bit implement Interpolatable

srimanachanta, 2025-03-20:
Useful for LED features like breath and wave where color is smoothly transitioned linearly between two fixed colors.

Ex:
```
  private Color solid(Section section, Color color) {
    if (color != null) {
      for (int i = section.start(); i < section.end(); i++) {
        buffer.setLED(i, color);
      }
    }
    return color;
  }

  private Color breath(Section section, Color c1, Color c2, double duration, double timestamp) {
    double x = ((timestamp % duration) / duration) * 2.0 * Math.PI;
    double ratio = (Math.sin(x) + 1.0) / 2.0;
    var color = interpolate(c1, c2, ratio);
    solid(section, color);
    return color;
  }

  private void wave(Section section, Color c1, Color c2, double cycleLength, double duration) {
    double x = (1 - ((Timer.getTimestamp() % duration) / duration)) * 2.0 * Math.PI;
    double xDiffPerLed = (2.0 * Math.PI) / cycleLength;
    for (int i = section.end() - 1; i >= section.start(); i--) {
      x += xDiffPerLed;
      double ratio = (Math.pow(Math.sin(x), waveExponent) + 1.0) / 2.0;
      if (Double.isNaN(ratio)) {
        ratio = (-Math.pow(Math.sin(x + Math.PI), waveExponent) + 1.0) / 2.0;
      }
      if (Double.isNaN(ratio)) {
        ratio = 0.5;
      }
      buffer.setLED(i, interpolate(c1, c2, ratio));
    }
  }

  private Color interpolate(Color c1, Color c2, double t) {
    return new Color(
        MathUtil.interpolate(c1.red, c2.red, t),
        MathUtil.interpolate(c1.green, c2.green, t),
        MathUtil.interpolate(c1.blue, c2.blue, t)
    );
  }
```

srimanachanta, 2025-03-20:
After looking into it more, this use case is handled already by [Color::lerpRGB](https://github.com/wpilibsuite/allwpilib/blob/1efaaefd78a634599b522fc2420c7283722554fb/wpilibj/src/main/java/edu/wpi/first/wpilibj/util/Color.java#L278). With that being said, maybe it should be modified to be interpolated based on the methods outlined in this:
> Here's a good writeup with interactive demos: https://www.alanzucconi.com/2016/01/06/colour-interpolation/

closed 2025-03-20 by srimanachanta

auscompgeek, 2025-03-20:
FWIW, looks like that blog post predates Oklab. https://evilmartians.com/chronicles/oklch-in-css-why-quit-rgb-hsl