wpilibsuite/allwpilib #7879: Installing latest WPILib version flags antivirus as ransomware

NotMePipe, 2025-03-24:
**Describe the bug**
Whenever I run the WPILib Installer for version 2025.3.2, my antivirus (Malwarebytes) flags it as ransomware and quarantines it, stopping the installation.

**To Reproduce**
Run the WPILib 2025.3.2 installer (with malwarebytes installed, seeing as I don't know if other antiviruses also do this)

**Expected behavior**
WPILib 2025.3.2 should install correctly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop:**
 - OS: Windows 11
 - Malwarebytes Premium (Version 5.2.8.173)

**Additional context**
This is the first version of WPILib to have this issue for me

sciencewhiz, 2025-03-28:
Did you report it as a false positive to malwarebytes?

NotMePipe, 2025-07-02:
I did not, but the issue seemed to randomly stop occurring, so I completely forgot about this bug report; sorry